serialport/node-serialport #1728: Problem with raspberry pi 3+ and Arduino on NodeJS

Sousana9617, 2018-11-24:
With a friend, we are trying to build a program on a raspberry pi 3 B+ that is using a Arduino to function. It's working on his computer but since I've tested it on raspberry I've encourtered many problems. This is the last problem I've with it, I've past many days searching on Internet how to fix it, but there's nothing. Could you help me ?

Here are the screenshot : Thank you :)
[image]
[image]

HipsterBrown, 2018-11-26:
@Sousana9617 Have you looked through the installation instructions for Serialport on Raspberry-Pi? https://serialport.io/docs/en/guide-installation#raspberry-pi-linux

From the screenshot, it appears the native bindings for node-serialport are not included in the electron build.

Sousana9617, 2018-12-01:
Hello I've resolved my problem. Yes serialport was not good installed. Here's the solution
Installer serialport :

npm install –save-dev electron-rebuild
npm install serialport
$(npm bin)/electron-rebuild
npm install
npm start

I think it's not clear that I've to launch electron rebuild, for installing serialport in the documentation.

HipsterBrown, 2018-12-02:
> I think it's not clear that I've to launch electron rebuild, for installing serialport in the documentation.

I'm glad you got it working. It's definitely a mixed case of the Electron and Raspberry PI installation instructions. Not sure how we would label the instructions for that but if you had a tutorial for us to share, we could add that to the docs. 

Happy hacking! :smile:

closed 2018-12-02 by HipsterBrown