serialport/node-serialport #1729: *more* MacOS /dev/cu vs. /dev/tty problems

jalbert10NJ, 2018-11-27:
SerialPort version: versions 5.x, 6.x, and 7.x, vs. old SerialPort version 4.0.7
Operating System and Hardware Platform: OSX (I am now using 10.13.6 High Sierra, but had this issue with 10.12 as well)

Summary of Problem

SerialPort version 4.0.7 has worked excellently for me on MacOS for multiple years now, connecting to my /dev/cu ports (for example, /dev/cu.usbmodem1411 ) just fine.  Since mid-2017, anytime I have been pushed to upgrade to v5, v6, or now v7 versions, I have run into the /dev/cu vs. /dev/tty problem, which was ostensibly "fixed" (which very unfortunately for me equals "broken something that previously worked") in 

   https://github.com/node-serialport/node-serialport/issues/1084
and
   https://github.com/node-serialport/node-serialport/issues/1070

Now, since I very sadly can no longer use Serialport v4.0.7 and am forced to upgrade (the dependencies of the other node packages that I need to use will no longer compile at all with Serialport v4.0.7, and they of course compile just fine with more recent versions, such as the most recent Serialport 7.1.0), I've had to upgrade to 7.1.0 (I also tried this a few months ago with 6.1.1).

The problem is that data only comes out of my /dev/cu ports, but never comes out of my /dev/tty ports.  Serialport now will only list the /dev/tty's and not the /dev/cu's, but data will only come out of the latter: If, just on the OS X unix command line, I do a

    cat /dev/cu.usbmodem1411 

then I can see data spewing out, but

    cat /dev/tty.usbmodem1411

shows nothing coming out.  But (as mentioned above) SerialPort will only list and connect with the /dev/tty ports and not the /dev/cu ports now.  Even if I try to explicitly force SerialPort to connect with the /dev/cu port, by explicitly doing:

           myPort = new serialport('/dev/cu.usbmodem1411', {
               lock: false,
               baudRate: 38400,
               // look for return and newline at the end of each data packet:
//               parser: serialport.parsers.readline("\n")
                parser: new serialport.parsers.Readline('\r\n')
           });
          myPort.on('data', sendSerialData);

instead of:

           myPort = new serialport(arduinoPortName, {
               baudRate: 38400,
               // look for return and newline at the end of each data packet:
//               parser: serialport.parsers.readline("\n")
                parser: new serialport.parsers.Readline('\r\n')
           });
          myPort.on('data', sendSerialData);

(where arduinoPortName is the '/dev/tty.usbmodem1411' name that is now returned by serialport.list), no data ever gets returned to me -- i.e. my sendSerialData function is never called.  

I am not sure why even explicitly providing the /dev/cu name per the above does not work for me with SerialPort versions v5, v6, and v7.  But as mentioned things always worked fine with SerialPort v4.0.7.

My apologies if anything is unclear in the above, or if I forgot to add anything critical, and thanks very much in advance for any help.

jalbert10NJ, 2018-11-28:
It turns out that explicitly doing myPort = new serialport('/dev/cu.usbmodem1411') *was* opening the port, but I was doing 

           myPort.on('open', showPortOpen);

and my showPortOpen function needed to be changed from:

    function showPortOpen() {
        console.log('Serial port open.  Data rate: ' + myPort.options.baudRate);
        console.log("");
    }

to

    function showPortOpen() {
        console.log('Serial port open.  Data rate: ' + myPort.baudRate);
        console.log("");
    }

as the former does not work and rather silently freezes in the post-4.0.7 api.

To summarize, I had to change the following 4.0.7 code:

    var serialport = require('serialport');

    serialport.list(function (err, ports) {
        ports.forEach(function(port) {
           if (port.comName.indexOf("cu.usbmodem") != -1) arduinoPortName = port.comName;
        });
    });

           myPort = new serialport(arduinoPortName, {
               baudRate: 38400,
               // look for return and newline at the end of each data packet:
               parser: serialport.parsers.readline("\n")
           });
           myPort.on('open', showPortOpen);
           myPort.on('data', sendSerialData);
           myPort.on('close', showPortClose);
           myPort.on('error', showError);

    function showPortOpen() {
        console.log('Serial port open.  Data rate: ' + myPort.options.baudRate);
        console.log("");
    }

to this code that now appears to work in 7.1.0:

    var serialport = require('serialport');

    serialport.list(function (err, ports) {
        ports.forEach(function(port) {
           if (port.comName.indexOf("tty.usbmodem") != -1) arduinoPortName = port.comName.replace("tty.usbmodem", "cu.usbmodem");
        });
    });

           myPort = new serialport(arduinoPortName, {
               lock: false,
               baudRate: 38400
            });
           var readline = serialport.parsers.Readline;
           var parser = myPort.pipe(new readline('/r/n'));
           myPort.on('open', showPortOpen);
           parser.on('data', sendSerialData);
           myPort.on('close', showPortClose);
           myPort.on('error', showError);

    function showPortOpen() {
        console.log('Serial port open.  Data rate: ' + myPort.baudRate);
        console.log("");
    }

jalbert10NJ, 2018-11-28:
yes, the code replacements in my comment above ( https://github.com/node-serialport/node-serialport/issues/1729#issuecomment-442302980 ) appear to work -- they string-replace the tty's with cu's.  (Note that the tty's still do not work for me -- no data comes out of them on my particular system.)

jalbert10NJ, 2018-11-29:
Various Arduinos, mostly Mega 2560's, but also Micro's and Uno's.  Yes, I certainly can run with the TTY and debug enabled -- but it will be a few days (I have some urgent things I need to take care of first).

jalbert10NJ, 2018-12-10:
Works now for me with tty.  Perhaps the problem I was having with all versions beyond 4.0.7 previously, after changing cu's to tty's, was just that I hadn't additionally changed myPort.options.baudRate to myPort.baudRate in my showPortOpen function -- see the code I was using above (and hadn't noticed that it had actually been dying there, after changing cu's to tty's).  Perils of changing anything in an API (without making _absolutely everything_ backwards-compatible for stick-in-the-mud users like me, even silly perverse things like that ;) ...

reconbot, 2018-12-10:
We had a weird lowercase all the options process for a while that caused a bunch of headaches internally. This was a few years ago.

We do have this line to give a specific error message!
https://github.com/node-serialport/node-serialport/blob/master/packages/stream/stream.js#L119

I'm glad you figured it out!

closed 2018-12-10 by reconbot