serialport/node-serialport #1733: node-serialport and vue?

piotrrydygier, 2018-12-02:
Can I use node-serialport in Vue application?
When I import serialport I get this error: 
`bindings.js?dfc1:17 Uncaught TypeError: Cannot read property 'node' of undefined
    at Object.eval (bindings.js?dfc1:17)
    at eval (bindings.js:176)
    at Object../node_modules/bindings/bindings.js (app.js:1150)
    at __webpack_require__ (app.js:768)
    at fn (app.js:131)
    at eval (main.js?56d7:1)
    at Module../src/main.js (app.js:2707)
    at __webpack_require__ (app.js:768)
    at fn (app.js:131)
    at Object.0 (app.js:2780)`

reconbot, 2018-12-02:
The bindings package cannot be webpacked

piotrrydygier, 2018-12-02:
What can I do to work this module in vue app?

reconbot, 2018-12-02:
I don't know, I've never used it. I do know it uses webpack, we do have a
webpack section in our docs.

hanfengcan, 2018-12-03:
@piotrrydygier Is it a electron application with vue. or only a vue application run in browser?

reconbot, 2018-12-03:
Huh, we should totally have a webpack section.

TimPietrusky, 2018-12-03:
The only way for me to fix the `Error: Could not locate the bindings file. Tried:` that comes when using webpack was to create a symlink to the actual bindings. 

`ln -s ../node_modules/@serialport/bindings/build/Release/ build/Release`

(assuming that you build into the `build` folder)

scottosaur, 2018-12-07:
Not the most specific answer, but I'm using serialport in a Vue/Electron app scaffolded with ElectronForge and it was a pretty painless experience (barring a few early compilation headaches that were largely the fault of bad permissions structures on my dev PC).

Aside from the occasional hiccup of leaving off a "this" here and there while building the Vue scripts, it has worked admirably. Even able to control multiple devices on different ports by having different Vue components open separate ports.

closed 2019-02-13 by stale[bot]