serialport/node-serialport #1735: Getting "Error Resource temporarily unavailable Cannot lockport" while I am trying to have dynamic port connection.

iamsush, 2018-12-03:
### Summary of Problem
Getting "Error Resource temporarily unavailable Cannot lockport" while I am trying to have dynamic port connection.
(Please answer all 3)

- What are you trying to do?
=> My requirement is to open a port dynamically (on the basis of two variables : {port: *to which I want to connect*} and {enable: *whether I want autoOpen port or not*})
- What happens?
=> Everything is working fine, and I am able to switch to a new port. But, when I trying to select a port which I have already used (and didn't close), I am getting the below error
 "Error Resource temporarily unavailable Cannot lockport"

PS: I want to create a singleton instance of Serialport  but as per the Error I guess I am not clearing out instances I have created before selecting a new instance. I am stuck with this for so many days. Please help. 

- What should have happened?
=> Singleton Instance of Serialport.

### Code to Reproduce the Issue

const SerialPort = require('serialport');
let isPortOpen = false;
let currentPort = '';
var hardwarePort;

const createPort = (port, enable) => {
    hardwarePort = new SerialPort(port, {
        baudRate: 115200,
        autoOpen: enable
    });
    currentPort = port;
    isPortOpen = enable;
    hardwarePort.on('data', function (recieved) {
        const arr = [...recieved];
        console.log(arr);
    });

    hardwarePort.on('open', function () {
        console.log('opened');
    });
    hardwarePort.on('close', function () {
        console.log('closed');
    });

    hardwarePort.on('error', err => {
        console.log(err);
    });
};

module.exports = {
    createPort: (port, enable) => {
        console.log(currentPort);
        console.log(port);
        if (port === currentPort) {
            console.log('using previous');
            return;
        }
        console.log('creating port');
        createPort(port, enable);
        return;
    },
    getInstance: () => {
        return hardwarePort;
    },
    closePort: () => {
        if (isPortOpen && hardwarePort !== undefined) {
            hardwarePort.close();
            isPortOpen = false;
        }
    },
    openPort: () => {
        if (!isPortOpen && hardwarePort !== undefined) {
            hardwarePort.open();
            isPortOpen = true;
        }
    }
};

### Versions, Operating System and Hardware

- SerialPort@?: "serialport": "^7.0.2"
- Node.js v8.14.0
- Linux
- Hardware and chipset? (Prolific/FTDI/Other): not sure

HipsterBrown, 2018-12-03:
@iamsush 

So here is what I would do to manage multiple Serialport instances:

Keep a [`Map`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Map) of instances in your module and close old instances before trying to recreate them:

```is
const createdPorts = new Map();

const createPort = (port, enable) => {
    if (createdPorts.has(port)) {
        const oldPort = createdPorts.get(port);
        // you could close the port before recreating it
        // or return the old instance as hardwarePort
        if (oldPort.isOpen) oldPort.close();
    }

    hardwarePort = new SerialPort(port, {
        baudRate: 115200,
        autoOpen: enable
    });
    currentPort = port;

    createdPorts.set(port, hardwarePort);

    // not needed when hardwarePort has an 'isOpen` property
    isPortOpen = enable;

    hardwarePort.on('data', function (recieved) {
        const arr = [...recieved];
        console.log(arr);
    });

    hardwarePort.on('open', function () {
        console.log('opened');
    });
    hardwarePort.on('close', function () {
        console.log('closed');
    });

    hardwarePort.on('error', err => {
        console.log(err);
    });
};

module.exports = {
    createPort: (port, enable) => {
        console.log(currentPort);
        console.log(port);
        if (port === currentPort) {
            console.log('using previous');
            return;
        }
        console.log('creating port');
        createPort(port, enable);
        return;
    },
    getInstance: () => {
        return createdPorts.get(currentPort);
    },
    closePort: () => {
        if (isPortOpen && hardwarePort !== undefined) {
            hardwarePort.close();
            isPortOpen = false;
        }
    },
    openPort: () => {
        if (!isPortOpen && hardwarePort !== undefined) {
            hardwarePort.open();
            isPortOpen = true;
        }
    }
};
```

I hope the makes sense. Let me know if that works out.

iamsush, 2018-12-03:
Thanks @HipsterBrown, 
This would be better than what I am trying to do right now. But I have few concerns : 
> I just want to have Single Instance of hardwarePort, but my code is referring to the previous instances (as far as I know ES6 modules create singleton instances).
(My understanding is, if I am creating a new reference to hardwarePort, the previous reference should be collected by garbage collector. But what I have seen is that, even after changing the reference the port is still open. I am not sure where I am going wrong in my understandings) 

Does the above implementation cause any memory leaks or any other drawbacks?

I guess that I am going wrong somewhere with the JavaScript basics and you might not be the concerned person. But please let me know, if you can help.

>Also, I am getting below error while using isOpen,
"TypeError: hardwarePort.isOpen is not a function"

HipsterBrown, 2018-12-04:
> if I am creating a new reference to hardwarePort, the previous reference should be collected by garbage collector

This may happen but it doesn't close the port if it does. There is still the native binding keeping it open until you close it. If you are switching ports, it's best to close all open ports before reassigning `hardwarePort` to avoid that "lockport" error. 

> Does the above implementation cause any memory leaks or any other drawbacks?

I'm not sure about any memory leaks that could cause. You probably don't have enough ports to cause out-of-memory errors from storing instances in a Map.

closed 2019-03-26 by stale[bot]