serialport/node-serialport #1736: Docker in Docker Build Fails

sirockin, 2018-12-04:
### Summary of Problem

- What are you trying to do?
Build a docker image of my application from a CI environment (Bitbucket Pipelines).

- What happens?
Whether I run `npm install` directly or run `npm install serialport --build-from-source`, I get the same error output:
```
npm ERR! code E404
npm ERR! 404 Not Found: event-stream@https://registry.npmjs.org/event-stream/-/event-stream-3.3.6.tgz
```

- What should have happened?
The package should have installed.

### Versions, Operating System and Hardware

- SerialPort@? 6.2.2
- Node.js v?8.9.3
- Windows? Linux? Mac? Linux (docker:dind)

reconbot, 2018-12-04:
That looks like the malicious version that got pulled from the registry.
You should upgrade.

sirockin, 2018-12-04:
Thanks for the feedback. I've just upgraded to 7.1.0 and got exactly the same error.

reconbot, 2018-12-05:
it's not a package serialport uses
---
Francis Gulotta
wizard@roborooter.com

sirockin, 2018-12-05:
Sorry my bad. I have a  a real problem building with serialport in Docker in Docker but this wasn't it. I'll close this and report the problem in a new issue.

closed 2018-12-05 by sirockin