serialport/node-serialport #1744: serialport does not work on windows 10

huydox, 2018-12-17:
### Summary of Problem
I tried to get an example of reading data from COM port on Windows but no luck. There is no data emit on data event of serialport as well as parser.

(Please answer all 3)

- What are you trying to do?
Just read data from serialport.
- What happens?
No data got in data event of serialport or parser. 
If I stop code as read function of library in debug mode, and make data available before continue, data will be emitted.
After that, `end` event will be emit and no data can be receive after that.
- What should have happened?
I should get data whenever data come to my COM port
### Code to Reproduce the Issue

```js
const SerialPort = require('serialport');
const Readline = require('@serialport/parser-readline');
const port = new SerialPort('COM1', { 
    baudRate: 9600,
    dataBits: 8,
    parity: 'none',
    stopBits: 1,
    highWaterMark: 32,
    hupcl: false
});

parser = port.pipe(new Readline({delimiter: '\n', encoding:'utf8'}));

parser.on('data', function(data){
  console.log('parser data:', data);
})

port.on('data', data =>{
  console.log('got word from arduino:', data);
  console.log(data.toString());
});

port.on('error', (err) => {
  console.log('port err', err);
});

port.on('close', () => {
  console.log('port close');
})

port.on('end', () => {
  console.log('port end');
})

port.on('open', function(){
  console.log('port already open');
})
```
### Output
```
server>node .\test.js
port already open
port end
server>
```
### Versions, Operating System and Hardware

- SerialPort@7.1
- Node.js v8.12
- Windows 10
- Hardware and chipset: ATC-1200 (Ethernet to Serial Converter)

huydox, 2018-12-18:
I believe that issue come bindings.node (which built from C++).
In work case, if there is no data, bytesRead always return undefined and keep waiting for data.
In failed case, bytesRead returns 0 and stop waiting for data.

HipsterBrown, 2018-12-18:
@huydox What are you using to create the virtual ports? Is the program called Virtual COM?

huydox, 2018-12-18:
Hi @HipsterBrown,
I'm using this one - which included with ATC-1200

[image]
 product

HipsterBrown, 2018-12-18:
@huydox Can you run `npx @serialport/list` and see if your virtual port is listed?

huydox, 2018-12-18:
Yes @HipsterBrown, list works and prints virtual port.
 I can also open virtual port success.
With my debug above by setting break point in read function, then send data to COM port, it can read data. However when it reads to EOF, there is a problem that cause the stream end and no longer wait for next data.
So I guest the problem some thing like read null data.

HipsterBrown, 2018-12-19:
Thanks for the update, that's great to know. Happy building! 🎉

closed 2018-12-19 by HipsterBrown