serialport/node-serialport #1747: Serial port hangs on Raspberry Pi 3

BryanHunt, 2018-12-20:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
I'm trying to read data from the serial port on a Raspberry Pi 3
- What happens?
Data is read successfully for the firs 20 or so data points and then hangs
- What should have happened?
Data should be read continuously
### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')
const Readline = require('@serialport/parser-readline')
const port = new SerialPort('/dev/serial0', { baudRate: 9600 })
const parser = new Readline({ delimiter: '\r'})
port.pipe(parser)
parser.on('data', console.log)
```

### Versions, Operating System and Hardware

- SerialPort@? 7.1.1
- Node.js v? v10.9.0
- Windows? Linux? Mac? Linux (DietPi)
- Hardware and chipset? (Prolific/FTDI/Other) Whatever is in the RaspberryPi 3

When running with debug enabled, the above app will terminate instead of hanging.  I have run the code on a different Raspberry Pi 3 with the same result and I've attached an oscilloscope to the serial port to see the hardware transmitting the serial data properly, so I think I've eliminated the hardware.  I'll try the same code running under Raspian next.  I'm attaching the debug log output.

[log.txt](https://github.com/node-serialport/node-serialport/files/2699546/log.txt)

BryanHunt, 2018-12-20:
Yes, I'm running `serialport` on the Pi and reading data from a device connected to it using `/dev/serial0`.  The device continuously streams data in the form `Rxxxx\r` where `xxxx` is a number.

I added `port.on('error', (error) => console.log(error))` to the code and got no additional output.

HipsterBrown, 2018-12-20:
What device are you using?

Can you also try listening for the `close` event? Just trying to check off any potential case around the "bad file descriptor" error.

BryanHunt, 2018-12-20:
The device connected to the Pi is an ultrasonic transducer: https://www.maxbotix.com/Ultrasonic_Sensors/MB7380.htm

I added the `close` event and didn't get any data.

BryanHunt, 2018-12-20:
I just tried `serialport 6.2.2` and get a slightly different result.  When I run without debug, instead of hanging, I get:

```
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
0300
0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
R0300
{ [Error: bad file descriptor] disconnected: true }
```

HipsterBrown, 2018-12-20:
Interesting, so the port is disconnecting. Not sure if that's an issue on the Pi or with Serialport. Do you have another platform you can try it on?

Sounds like we also have an issue with surfacing disconnect errors in the latest versions.

BryanHunt, 2018-12-21:
I reproduced the fail with Raspbian and then I got it working.

To enable the serial port on Raspbian, you launch `raspi-config`, then select `Interfacing Options`, then `Serial`.  You will then be asked two questions:

1.  Would you like a login shell to be accessible over serial?
2. Would you like the serial port hardware to be enabled?

The first time I tried it, I mistakenly answered `yes` to both questions and that configuration exactly reproduces the problem.

If you answer `no` to question 1 and `yes` to question 2, then the code and serial port work perfectly.

This makes sense as you don't want a login shell attached to your serial port when you are trying to use it as a serial port.

`dietpi-config` has a similar option to enable the serial port, except that it doesn't ask you if you want the login console or not.  I'm still trying to figure out how to enable the serial port without the login console on DietPi.

reconbot, 2018-12-21:
Adding this to our docs would be amazing

HipsterBrown, 2018-12-21:
@BryanHunt Great work! :clap:

Could you add your findings in a PR to this doc -> https://github.com/node-serialport/node-serialport/blob/master/docs/guide-installation.md#raspberry-pi-linux

closed 2019-02-13 by reconbot