serialport/node-serialport #1751: receiving data delayed

Holger-Will, 2018-12-21:
## prelude

i have a problem which is a bit hard to explain. but i'll try:

i have a simple express+socket.io server which acts as a serialport to websocket gateway.
it is part of a lib i'm developing: https://github.com/enocean-js/enocean-js

the server code can be found here: https://github.com/enocean-js/enocean-js/blob/master/packages/node_modules/%40enocean-js/websocket-gateway/server.js

if you clone the repo, and `npm i`, everything works as expected...
but when i remove my package.lock and then `npm i` the Bug show up.

i'm using my own parser: https://github.com/enocean-js/enocean-js/blob/master/packages/node_modules/%40enocean-js/serialport-parser/src/serialport-parser.js

## Bug

when receiving data on the serialport, the parser.on("data",...) event does not fire when the parser finished parsing the input, but after a few seconds, all detected packets/events are emitted. 

in the meantime i can still write to the port without a problem.

now here comes the part i hope will help you find the problem: when i insert a timer

    setInterval(()=>{},0)

everthing works again as expected... moreover, if i set the interval to any  value, the events are fired at exactly that interval...

i hope this gives you a hint at where the problem might be...

## p.s.

i'm on linux (Arch)
tried with different node versions (my min-version is 8.4.0) up to 11.4.0.

mhmtsrfglu, 2018-12-22:
`setInterval(()=>{},0)`
It solved my problem. Thanks for solution

reconbot, 2018-12-22:
Can you run npm ls before and after removing your package-lock.json? I would like to know if the serialport version changes

Holger-Will, 2018-12-22:
before: 
serialport@7.1.0
│ ├─┬ @serialport/binding-mock@2.0.2
│ │ ├─┬ @serialport/binding-abstract@2.0.2
│ │ │ └─┬ debug@4.1.0
│ │ │   └── ms@2.1.1
│ │ └─┬ debug@4.1.0
│ │   └── ms@2.1.1
│ ├─┬ @serialport/bindings@2.0.3
│ │ ├── @serialport/binding-abstract@2.0.2 deduped
│ │ ├── @serialport/parser-readline@2.0.2 deduped
│ │ ├── bindings@1.3.1
│ │ ├─┬ debug@4.1.0
│ │ │ └── ms@2.1.1
│ │ ├── nan@2.11.1
│ │ └─┬ prebuild-install@5.2.2
│ │   ├── detect-libc@1.0.3
│ │   ├── expand-template@2.0.3
│ │   ├── github-from-package@0.0.0
│ │   ├── minimist@1.2.0
│ │   ├── mkdirp@0.5.1 deduped
│ │   ├── napi-build-utils@1.0.1
│ │   ├─┬ node-abi@2.5.0
│ │   │ └── semver@5.6.0 deduped
│ │   ├── noop-logger@0.1.1
│ │   ├─┬ npmlog@4.1.2
│ │   │ ├─┬ are-we-there-yet@1.1.5
│ │   │ │ ├── delegates@1.0.0
│ │   │ │ └─┬ readable-stream@2.3.6
│ │   │ │   ├── core-util-is@1.0.2
│ │   │ │   ├── inherits@2.0.3 deduped
│ │   │ │   ├── isarray@1.0.0 deduped
│ │   │ │   ├── process-nextick-args@2.0.0
│ │   │ │   ├── safe-buffer@5.1.2 deduped
│ │   │ │   ├─┬ string_decoder@1.1.1
│ │   │ │   │ └── safe-buffer@5.1.2 deduped
│ │   │ │   └── util-deprecate@1.0.2
│ │   │ ├── console-control-strings@1.1.0
│ │   │ ├─┬ gauge@2.7.4
│ │   │ │ ├── aproba@1.2.0
│ │   │ │ ├── console-control-strings@1.1.0 deduped
│ │   │ │ ├── has-unicode@2.0.1
│ │   │ │ ├── object-assign@4.1.1 deduped
│ │   │ │ ├── signal-exit@3.0.2
│ │   │ │ ├─┬ string-width@1.0.2
│ │   │ │ │ ├── code-point-at@1.1.0
│ │   │ │ │ ├─┬ is-fullwidth-code-point@1.0.0
│ │   │ │ │ │ └── number-is-nan@1.0.1
│ │   │ │ │ └── strip-ansi@3.0.1 deduped
│ │   │ │ ├─┬ strip-ansi@3.0.1
│ │   │ │ │ └── ansi-regex@2.1.1 deduped
│ │   │ │ └─┬ wide-align@1.1.3
│ │   │ │   └── string-width@2.1.1 deduped
│ │   │ └── set-blocking@2.0.0
│ │   ├── os-homedir@1.0.2
│ │   ├─┬ pump@2.0.1
│ │   │ ├─┬ end-of-stream@1.4.1
│ │   │ │ └── once@1.4.0 deduped
│ │   │ └── once@1.4.0 deduped
│ │   ├─┬ rc@1.2.8
│ │   │ ├── deep-extend@0.6.0
│ │   │ ├── ini@1.3.5
│ │   │ ├── minimist@1.2.0
│ │   │ └── strip-json-comments@2.0.1 deduped
│ │   ├─┬ simple-get@2.8.1
│ │   │ ├─┬ decompress-response@3.3.0
│ │   │ │ └── mimic-response@1.0.1
│ │   │ ├── once@1.4.0 deduped
│ │   │ └── simple-concat@1.0.0
│ │   ├─┬ tar-fs@1.16.3
│ │   │ ├── chownr@1.1.1
│ │   │ ├── mkdirp@0.5.1 deduped
│ │   │ ├─┬ pump@1.0.3
│ │   │ │ ├── end-of-stream@1.4.1 deduped
│ │   │ │ └── once@1.4.0 deduped
│ │   │ └─┬ tar-stream@1.6.2
│ │   │   ├─┬ bl@1.2.2
│ │   │   │ ├── readable-stream@2.3.6 deduped
│ │   │   │ └── safe-buffer@5.1.2 deduped
│ │   │   ├─┬ buffer-alloc@1.2.0
│ │   │   │ ├── buffer-alloc-unsafe@1.1.0
│ │   │   │ └── buffer-fill@1.0.0
│ │   │   ├── end-of-stream@1.4.1 deduped
│ │   │   ├── fs-constants@1.0.0
│ │   │   ├── readable-stream@2.3.6 deduped
│ │   │   ├── to-buffer@1.1.1
│ │   │   └── xtend@4.0.1 deduped
│ │   ├─┬ tunnel-agent@0.6.0
│ │   │ └── safe-buffer@5.1.2 deduped
│ │   └── which-pm-runs@1.0.0
│ ├── @serialport/parser-byte-length@2.0.2
│ ├── @serialport/parser-cctalk@2.0.2
│ ├── @serialport/parser-delimiter@2.0.2
│ ├─┬ @serialport/parser-readline@2.0.2
│ │ └── @serialport/parser-delimiter@2.0.2 deduped
│ ├── @serialport/parser-ready@2.0.2
│ ├── @serialport/parser-regex@2.0.2
│ ├─┬ @serialport/stream@2.0.2
│ │ ├── @serialport/binding-mock@2.0.2 deduped
│ │ └─┬ debug@4.1.0
│ │   └── ms@2.1.1
│ └─┬ debug@4.1.0
│   └── ms@2.1.1

Holger-Will, 2018-12-22:
after:
├─┬ serialport@7.1.1
│ ├─┬ @serialport/binding-mock@2.0.3
│ │ ├─┬ @serialport/binding-abstract@2.0.3
│ │ │ └─┬ debug@4.1.1
│ │ │   └── ms@2.1.1
│ │ └─┬ debug@4.1.1
│ │   └── ms@2.1.1
│ ├─┬ @serialport/bindings@2.0.4
│ │ ├── @serialport/binding-abstract@2.0.3 deduped
│ │ ├── @serialport/parser-readline@2.0.2 deduped
│ │ ├── bindings@1.3.1
│ │ ├─┬ debug@4.1.1
│ │ │ └── ms@2.1.1
│ │ ├── nan@2.12.1
│ │ └─┬ prebuild-install@5.2.2
│ │   ├── detect-libc@1.0.3
│ │   ├── expand-template@2.0.3
│ │   ├── github-from-package@0.0.0
│ │   ├── minimist@1.2.0
│ │   ├── mkdirp@0.5.1 deduped
│ │   ├── napi-build-utils@1.0.1
│ │   ├─┬ node-abi@2.5.1
│ │   │ └── semver@5.6.0 deduped
│ │   ├── noop-logger@0.1.1
│ │   ├─┬ npmlog@4.1.2
│ │   │ ├─┬ are-we-there-yet@1.1.5
│ │   │ │ ├── delegates@1.0.0
│ │   │ │ └─┬ readable-stream@2.3.6
│ │   │ │   ├── core-util-is@1.0.2
│ │   │ │   ├── inherits@2.0.3 deduped
│ │   │ │   ├── isarray@1.0.0 deduped
│ │   │ │   ├── process-nextick-args@2.0.0
│ │   │ │   ├── safe-buffer@5.1.2 deduped
│ │   │ │   ├─┬ string_decoder@1.1.1
│ │   │ │   │ └── safe-buffer@5.1.2 deduped
│ │   │ │   └── util-deprecate@1.0.2
│ │   │ ├── console-control-strings@1.1.0
│ │   │ ├─┬ gauge@2.7.4
│ │   │ │ ├── aproba@1.2.0
│ │   │ │ ├── console-control-strings@1.1.0 deduped
│ │   │ │ ├── has-unicode@2.0.1
│ │   │ │ ├── object-assign@4.1.1 deduped
│ │   │ │ ├── signal-exit@3.0.2
│ │   │ │ ├─┬ string-width@1.0.2
│ │   │ │ │ ├── code-point-at@1.1.0
│ │   │ │ │ ├─┬ is-fullwidth-code-point@1.0.0
│ │   │ │ │ │ └── number-is-nan@1.0.1
│ │   │ │ │ └── strip-ansi@3.0.1 deduped
│ │   │ │ ├── strip-ansi@3.0.1 deduped
│ │   │ │ └─┬ wide-align@1.1.3
│ │   │ │   └── string-width@1.0.2 deduped
│ │   │ └── set-blocking@2.0.0
│ │   ├── os-homedir@1.0.2
│ │   ├─┬ pump@2.0.1
│ │   │ ├─┬ end-of-stream@1.4.1
│ │   │ │ └── once@1.4.0 deduped
│ │   │ └── once@1.4.0 deduped
│ │   ├─┬ rc@1.2.8
│ │   │ ├── deep-extend@0.6.0
│ │   │ ├── ini@1.3.5
│ │   │ ├── minimist@1.2.0
│ │   │ └── strip-json-comments@2.0.1 deduped
│ │   ├─┬ simple-get@2.8.1
│ │   │ ├─┬ decompress-response@3.3.0
│ │   │ │ └── mimic-response@1.0.1
│ │   │ ├── once@1.4.0 deduped
│ │   │ └── simple-concat@1.0.0
│ │   ├─┬ tar-fs@1.16.3
│ │   │ ├── chownr@1.1.1
│ │   │ ├── mkdirp@0.5.1 deduped
│ │   │ ├─┬ pump@1.0.3
│ │   │ │ ├── end-of-stream@1.4.1 deduped
│ │   │ │ └── once@1.4.0 deduped
│ │   │ └─┬ tar-stream@1.6.2
│ │   │   ├─┬ bl@1.2.2
│ │   │   │ ├── readable-stream@2.3.6 deduped
│ │   │   │ └── safe-buffer@5.1.2 deduped
│ │   │   ├─┬ buffer-alloc@1.2.0
│ │   │   │ ├── buffer-alloc-unsafe@1.1.0
│ │   │   │ └── buffer-fill@1.0.0
│ │   │   ├── end-of-stream@1.4.1 deduped
│ │   │   ├── fs-constants@1.0.0
│ │   │   ├── readable-stream@2.3.6 deduped
│ │   │   ├── to-buffer@1.1.1
│ │   │   └── xtend@4.0.1 deduped
│ │   ├─┬ tunnel-agent@0.6.0
│ │   │ └── safe-buffer@5.1.2 deduped
│ │   └── which-pm-runs@1.0.0
│ ├── @serialport/parser-byte-length@2.0.2
│ ├── @serialport/parser-cctalk@2.0.2
│ ├── @serialport/parser-delimiter@2.0.2
│ ├─┬ @serialport/parser-readline@2.0.2
│ │ └── @serialport/parser-delimiter@2.0.2 deduped
│ ├── @serialport/parser-ready@2.0.2
│ ├── @serialport/parser-regex@2.0.2
│ ├─┬ @serialport/stream@2.0.3
│ │ ├── @serialport/binding-mock@2.0.3 deduped
│ │ └─┬ debug@4.1.1
│ │   └── ms@2.1.1
│ └─┬ debug@4.1.1
│   └── ms@2.1.1

reconbot, 2018-12-22:
Can you test a simple setup, maybe a console log, with the two versions?

Holger-Will, 2018-12-22:
I will try to make a minimal test case. For now I was only able to reproduce the bug with serialport + websocket. Though it seems it does not depend on the websocket implementation. I tried ws, websocket, socket.io... I'm still not sure if a mock binding shows the same problem or if I have to get real data... What is the preferred data/parser for these minimal test cases?

Holger-Will, 2018-12-22:
I could make an arduino sending random bytes to the port... That should be the easiest for everyone to reproduce...

Holger-Will, 2018-12-23:
i was able to reproduce the bug without any third party modules. it's quite easy for everyone to test. 

    const SerialPort = require('serialport')
    const Readline = require('@serialport/parser-readline')

    const port = new SerialPort('/dev/ttyACM0', { baudRate: 9600 })
    const parser = port.pipe(new Readline())

    parser.on('data', console.log)

    setInterval(() => {}, 2000)

without the last line everything is fine. but when inserting the interval, data only comes in at the speed of the interval...

a complete setup including a simple arduino sketch can be cloned here: https://github.com/enocean-js/serialport-bug

Holger-Will, 2018-12-23:
there is a related report on SO : 
https://stackoverflow.com/questions/53892677/what-is-asynchronous-processing-when-using-settimeout

shabunin, 2018-12-25:
Can confirm the issue. Tried with different nodejs versions, all the same.

serialport@6.2.2 works good, as expected. 7.0.2, 7.1.1 has this issue.

Holger-Will, thanks for interval hack.

maxwellhadley, 2019-01-03:
I see the same issue with serialport 7.1.1 under NodeJS 8.12.0 on macOS 10.13.6 (High Sierra). It's as if the code that dispatches the received bytes into the JS environment is using the 'timers' thread, not the 'next tick' thread (or whatever the mechanism actually is).

reconbot, 2019-01-03:
Yeah this is related to https://github.com/node-serialport/node-serialport/issues/1742 we will either release a fix or pull the release.

closed 2019-01-06 by reconbot

Holger-Will, 2019-01-10:
i tried to confirm the fix, but now i'm getting a segfault with binding 2.0.5 whenever i try to open a port. probably a separate issue...

PR #1765 (2019-01-06): fix: propagate async context in callbacks
Fixes #1751 

I've tested this on Windows with real hardware (repro'ed the issue in #1751). I can't get [WSL](https://docs.microsoft.com/en-us/windows/wsl/faq) to work with a serial port right now to test on Ubuntu.

I can't think of a good way to make this commit smaller. `AsyncResource` needs to be constructed with a `const char*` name unique to each of your `struct`s (so can't inherit from some other class that extends `AsyncResource`). Constructing the `AsyncResource` (in a ctor or initializer list) makes it necessary to initialize everything else  [... cut]
commit 9b5dbdb2 (2019-01-06): fix: propagate async context in callbacks (#1765)

Fixes #1751