serialport/node-serialport #1752: Parser is not working

mhmtsrfglu, 2018-12-21:
### Summary of Problem

- I am trying to read data from my RFID device. When I try to use parser, I couldnt get data.

### Code to Reproduce the Issue

```js
var serialPort = require('serialport');

var serOptions = {
  baudRate: 9600,
  databits: 8,
  stopbits: 1,
  parity: 'none',
  buffersize: 4096,
  encoding: 'utf8'
};

const PORT = "COM6";
const port = new serialPort(PORT, serOptions);

port.on('error', function(err) {
    console.log('Error: ', err.message)
  })

port.on("data",(data)=>{
  console.log(data.toString("UTF8"));
});
// Code
```
this code is working but generating multiple buffer and it is not useful for me. It is generating
F
OO
D1
 
but I need FOOD1.

```js
const SerialPort = require('serialport')
const Readline = SerialPort.parsers.Readline;

myport = new SerialPort("COM6", {
    baudRate: 9600,
    databits: 16,
    parity: 'none',
    stopBits: 1,
    buffersize : 4096,
    flowControl: true,
    }
  );

 const parser = new Readline('\n\r');

 myport.pipe(parser);
 parser.on('data', (data) => {
 console.log(data)})
// Code
```
this code is with parser but it is not generating anything.

What is my mistake ?

reconbot, 2018-12-21:
Are you sure you have the right delimiter? Windows line endings are `\r\n` and unix ones are just `\n`

mhmtsrfglu, 2018-12-21:
> Are you sure you have the right delimiter? Windows line endings are `\r\n` and unix ones are just `\n`

I tried by one by for all, i couldnt get a result from anyone. 

` const parser = new Readline('\n');`
` const parser = new Readline('\r');`
` const parser = new Readline();`

I tried all but I still have same result

reconbot, 2018-12-21:
If you dump the raw data in HEX format it makes it easier to read what delimiter your device is sending.

```
port.on("data",(data)=>{
  console.log(data.toString("HEX"));
});
```

mhmtsrfglu, 2018-12-21:
> port.on("data",(data)=>{ console.log(data.toString("HEX")); });

my device sending data that below
46
4f4f4431 for FOOD1

reconbot, 2018-12-21:
`464f4f4431` is exactly `FOOD1` there's no delimiters to parse, `\n` is not in that data.

mhmtsrfglu, 2018-12-21:
> ```
> > Buffer.from('FOOD1').toString('HEX')
> '464f4f4431'
> ```
> `464f4f4431` is exactly `FOOD1` there's no delimiters to parse, `\n` is not in that data. A delimiter are byes that signify the end of a message.
> 
> If it sent `FOOD1\n` you could use the readline parser without issue. If every message is the same length you could use the byte length parser

It worked for me. Thank for solution! :)

closed 2018-12-21 by mhmtsrfglu