serialport/node-serialport #1755: Node module version conflict?

Dentrax, 2018-12-25:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?

Trying to install `serialport` for MagicMirror.

```
WARNING! Could not load config file. Starting with default configuration. Error found: Error: The module '/home/dentrax/Projects/GitHub/MagicMirror/modules/MMM-Test/node_modules/serialport/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 67. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

### Code to Reproduce the Issue

In my `package.json` file :

```json
    "dependencies": {
        "serialport": "^7.1.1"
    }
```

### Versions, Operating System and Hardware

- SerialPort@? `v7.1.1`
- Node.js v? `v11.5.0`
- npm v? `v6.5.0`
- Windows? Linux? Mac? `ArchLinux`
- Hardware and chipset? (Prolific/FTDI/Other) `Intel x86-64`

reconbot, 2018-12-26:
I don't know much about magic mirror, do you do a normal npm install? Is the node that runs your code the same as the node npm uses? does it use electron?

reconbot, 2018-12-26:
Unrelated to your error message, we probably don't support node 12 yet, unsure about 11

Dentrax, 2018-12-26:
@reconbot My full error:

```
Error: Module did not self-register.
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at bindings (/home/dentrax/Projects/GitHub/MagicMirror/modules/MMM-Test/node_modules/bindings/bindings.js:84:48)
    at Object.<anonymous> (/home/dentrax/Projects/GitHub/MagicMirror/modules/MMM-Test/node_modules/@serialport/bindings/lib/linux.js:1:191)
Whoops! There was an uncaught exception...
Error: Module did not self-register.
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at bindings (/home/dentrax/Projects/GitHub/MagicMirror/modules/MMM-Test/node_modules/bindings/bindings.js:84:48)
    at Object.<anonymous> (/home/dentrax/Projects/GitHub/MagicMirror/modules/MMM-Test/node_modules/@serialport/bindings/lib/linux.js:1:191)
```

Code:

```
var NodeHelper = require("node_helper");
var request = require('serialport');
var Readline = require('@serialport/parser-readline');

module.exports = NodeHelper.create({

    start: function() {
        console.log("=========== Starting node_helper for module [" + this.name + "]");
        var port = new SerialPort('/dev/ttyUSB0', { baudRate: 256000 });
        const parser = new Readline();
        port.pipe(parser);

        port.on("open", function () {
            console.log('open');
            port.on('data', function(data) {
                console.log(data);
            });
        });
    },
});
```

reconbot, 2018-12-26:
That does look like electron, Electron uses a different nodejs than npm, so when the binaries get installed you get the wrong one. You can look at our docs on electron and see that helps.

https://serialport.io/docs/en/guide-installation#electron

chiamtc, 2019-03-05:
Can confirm this is electron not working well with the latest version of Node Js. I was using Nodejs v11, electron 4.0.5 (which changes to NODE_MODULE_VERSION 69 based on their release note) and serialport 7.1.4.

My solution is to downgrade to Nodejs v10, electron to <4.0.4. It solves my problem with other libraries like grpc and node-sass as well.

closed 2019-03-17 by Dentrax