serialport/node-serialport #1757: File not found

DoctorVWA, 2018-12-28:
### Summary of Problem

The program can't open the COM port, it's located at id 1 (`COM1`), it returns `Error: Opening COM1: File not found`, I have tried to run with multiple paths like `\\.\COM1`, `\Device\COM1`, `\Device\Serial2` and a lot of random things related to regedit paths and random paths that I found on the internet. Also I tried to list all of the ports and then open the first one, that I saw in one example on the internet. My main objetive is just recieve data from the Arduino at COM port 1.

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport');

SerialPort.list((err, ports) => {
  if (err) {
    return console.log(err);
  }

  ports.forEach((path) => {
    console.log(path);

    let { comName } = path;

    const port = new SerialPort(comName, {
      baudRate: 9600
    }, console.log);

    port.on('data', console.log);
  });
});
```

### Log

```
{ comName: 'COM1',
  manufacturer: 'wch.cn',
  serialNumber: '5&19ab1ab1&0&1',
  pnpId: 'USB\\VID_1A86&PID_7523\\5&19AB1AB1&0&1',
  locationId: 'Port_#0001.Hub_#0006',
  vendorId: '1A86',
  productId: '7523' }
Error: Opening COM1: File not found
```

### Versions, Operating System and Hardware

- SerialPort@^7.1.1
- Node.js v7.10.0
- Windows 7 Ultimate 64 bits SP1
- Hardware and chipset? Arduino Uno R3 Smd

HipsterBrown, 2019-01-03:
@DoctorVWA Have you tried using a previous version of Serialport? Can you try installing `serialport@7.1.0` to see if we have a regression in the current patch?

Your example code looks fine to me. I don't have a Windows machine to test on.

DoctorVWA, 2019-01-03:
Yes, that worked, thanks

HipsterBrown, 2019-01-03:
Ok, thanks for letting us know.

@reconbot Here is another regression example. I can work on reverting the last release and publishing a new patched version tomorrow.

HipsterBrown, 2019-01-08:
@DoctorVWA `serialport@7.1.2` has been published. Can you let us know if that version works for you?

DoctorVWA, 2019-01-09:
Yes, that's working

HipsterBrown, 2019-01-09:
Awesome! Happy coding! 😄

closed 2019-01-09 by HipsterBrown

ananya-g, 2019-03-27:
i cant make it work on linux 18.04.
serialport version: 7.1.4