serialport/node-serialport #1760: Error: Method Not Allowed

Rajanenovate, 2019-01-02:
**During installation I am phasing this issue**

$ npm i serialport
(node:1716) [DEP0022] DeprecationWarning: os.tmpDir() is deprecated. Use os.tmpdir() instead.
npm WARN package.json node-cmd@3.0.0 license should be a valid SPDX license expression
npm ERR! Error: Method Not Allowed
npm ERR!     at errorResponse (/usr/share/npm/lib/cache/add-named.js:260:10)
npm ERR!     at /usr/share/npm/lib/cache/add-named.js:203:12
npm ERR!     at saved (/usr/share/npm/node_modules/npm-registry-client/lib/get.js:167:7)
npm ERR!     at FSReqWrap.oncomplete (fs.js:135:15)
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.14.79-v7+
npm ERR! command "/usr/bin/node" "/usr/bin/npm" "i" "serialport"
npm ERR! cwd /home/pi/js_programs/temp_js_serial
npm ERR! node -v v8.11.1
npm ERR! npm -v 1.4.21
npm ERR! code E405
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/js_programs/temp_js_serial/npm-debug.log
npm ERR! not ok code 0

reconbot, 2019-01-02:
You're using npm v1.4.21? Thanks incredibly old. I would upgrade.

HipsterBrown, 2019-01-03:
This looks like a specific issue with using an old npm version with newer versions of Node, especially as the stack trace shows the error coming from the npm cli. Closing as it is unrelated to this repo.

closed 2019-01-03 by HipsterBrown

Rajanenovate, 2019-01-04:
Yeah, after updating NPM to latest version problem solved.
Thanks