serialport/node-serialport #1761: net.createServer & serialport

jc-59, 2019-01-02:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
*receive data from SerialPort and print them on Console
*receive data from http request (port 8019) and send them to SerialPort
- What happens?
SerialPort is not opened

- What should have happened?
*SerialPort should open & data should be received from SerailPort

the following code is working with @serialport/bindings:2.0.3 but not anymore with @serialport/bindings:2.0.4 
### Code to Reproduce the Issue
var server = net.createServer(function(c) {
                        console.log((new Date()) + " - Server connected");
                        c.on('error', function(e) {
                                console.log((new Date()) + " - Error server disconnected");
                        });
                        c.on('close', function() {
                                console.log((new Date()) + " - Connexion closed");
                        });
                        c.on('data', function(data) {
                                console.log((new Date()) + " - Response: " + data);
                                gw.write(data.toString() + '\n');
                        });
                });
                server.listen(8019, function(e) {
                        console.log((new Date()) + " - server bound on 8019");
                });
        });

        var SerialPort = require('serialport');
        gw = new SerialPort(gwAddress);
        gw.settings.baudRate=115200;

        gw.on('open', function() {
                console.log((new Date()) + " - connected to serial);
        }).on('data', function(rd) {
                console.log((new Date()) + " - data received);

### Versions, Operating System and Hardware

- SerialPort@? 7.1.1
@serialport/bindings:2.0.4 (works well with @serialport/bindings:2.0.3)
- Node.js  v8.15.0
- Linux 
- Hardware and chipset? armv7l (Raspberry PI 3+)

floriansegginger, 2019-01-03:
I am having the same issue. Program just hangs, never opens the serial port. Even if the file descriptor does not exist.

Rolled back to @serialport/bindings@2.0.3 and everything works.

reconbot, 2019-01-03:
Yeah Sorry about this. This is a duplicate of #1751. We'll either pull the release or release an update.

closed 2019-01-03 by reconbot