serialport/node-serialport #1763: Write package size

JBtje, 2019-01-03:
### Summary of Problem
The device to which I write, has a buffer size of only 64 Bytes. It is possible to write unlimited data to it, but the packages cannot be bigger than 64 Bytes.
In my C# application, I added some code that divides the packages to 64 Bytes and sleep so that the device can clear the buffer for the next data.

Question: Is such a method implemented already?

### Code to Reproduce the Issue
**C#** code I use to write chunks of data to the serialport
```C#
        /// <summary>
        /// Write data to the COM Port basestream.
        /// </summary>
        /// <param name="data">Byte array: data to send.</param>
        private async Task Write(byte[] data)
        {
            var loops = (int)Math.Ceiling((double)data.Length / DeviceBufferLength);

            for (var iteration = 0; iteration < loops; iteration++)
            {
                // Take the Device Buffer Length, or the remainder length.
                var size = (iteration * DeviceBufferLength) + DeviceBufferLength > data.Length ? data.Length - (iteration * DeviceBufferLength) : DeviceBufferLength;

                await port.BaseStream.WriteAsync(data, iteration * DeviceBufferLength, size);

                if (size == DeviceBufferLength)
                {
                    Thread.Sleep(1);
                }
            }
        }
```

### Versions, Operating System and Hardware
- SerialPort@latest
- Node.js@latest
- Windows

HipsterBrown, 2019-01-03:
@JBtje If you check out the Node.js [Buffer documentation](https://nodejs.org/dist/latest-v10.x/docs/api/buffer.html), you can see methods like [`#byteLength`](https://nodejs.org/dist/latest-v10.x/docs/api/buffer.html#buffer_class_method_buffer_bytelength_string_encoding) and [`#alloc`](https://nodejs.org/dist/latest-v10.x/docs/api/buffer.html#buffer_class_method_buffer_alloc_size_fill_encoding) that can help you manage the size of the data you're sending to your device.

JBtje, 2019-01-10:
Here is a code snippet, for anyone who stumbles upon this problem.
It assumes you use @serialport/bindings directly (code here https://github.com/node-serialport/node-serialport/issues/1758#issuecomment-451298427), w/o the serialport wrapper.

```js
        async read( numBytes ) {
            let pkg = Buffer.alloc( numBytes );

            // console.log( 'read', 'Trying to read ' + numBytes + ' Bytes' );
            let ofs = 0;
            // Read the package, do so untill all bytes are red.
            while( ofs < numBytes ) {
                ofs += await this.SerialPort.read( pkg, ofs, numBytes - ofs );
            }
            // console.log( 'read', 'Done reading ' + numBytes + ' Bytes' );

            return pkg;
        },
```

closed 2019-01-10 by JBtje