serialport/node-serialport #1770: Opening a port does not work anymore with versions > 7.0.1 (MacOS & Ubuntu)

pierreroth64, 2019-01-09:
Opening a port does not work anymore on my MacOS with versions > 7.0.1.

I created a repo with all the informations: https://github.com/pierreroth64/serialport-test

Do not hesitate to ask for more details if needed.

Thanks for your help

HipsterBrown, 2019-01-09:
@pierreroth64 Have you tried the latest release of `serialport@7.1.2` to see if the issue is resolved there?

kjhe11e, 2019-01-09:
I am seeing a segfault when trying to open port with the following:
* serialport@7.1.2
* Node v10.8.0
* Ubuntu 18.04 LTS.

Segfault on trying `serialport@7.0.1` as well.
Same code works fine using `serialport@6.2.2`.

HipsterBrown, 2019-01-09:
@kjhe11e Is this with any port or a specific device?

kjhe11e, 2019-01-09:
I only have access to one specific hardware device at the moment. It automatically connects to `ttyUSB0`.
Will test with some different hardware as soon as I can, hopefully in next few days.

vrst37, 2019-01-09:
We can confirm the issue with Arduino (/usb/ttyACM0) serial port on Ubuntu 18.04 with nodejs 10.14.1 serial port module version 7.1.2 and 7.0.2

Holger-Will, 2019-01-10:
using https://github.com/ddopson/node-segfault-handler i get:

```
/serialport/node_modules/@serialport/bindings/build/Release/bindings.node(_Z5setupiP9OpenBaton+0x205)[0x7f631dcc9e15]
/serialport/node_modules/@serialport/bindings/build/Release/bindings.node(_Z8EIO_OpenP9uv_work_s+0x38)[0x7f631dcca078]

```

does that mean anything to you?

pierreroth64, 2019-01-10:
> @pierreroth64 Have you tried the latest release of `serialport@7.1.2` to see if the issue is resolved there?

Hi, sorry for the delayed answer. Yes, it still does not work with `serialport@7.1.2`. Tried on **several** ports on my MacBook, for two various connected devices => calling port.open() results in a segmentation fault.
For all of those hardware configurations (ports and connected devices), `serialport@6.2.2` works fine.

gganebnyi, 2019-01-10:
Same issue with 7.1.2. Latest MacOS + PL2303 TTL-to-USB adapter. 6.2.2 works fine.

reconbot, 2019-01-10:
I've pulled the bad releases, sorry for issues everyone. It's been hard to test with hardware on all the platforms, so I really thank you for these bug reports.

reconbot, 2019-01-10:
Does 7.1.0 work for everyone?

JBtje, 2019-01-10:
I had this problem this morning, and set @serialport/bindings to 2.0.3, which fixed the problem. If serialport 7.1.0 uses @serialport/bindings 2.0.3, it should work...

Holger-Will, 2019-01-10:
7.1.0 still segfaults for me :-(

Holger-Will, 2019-01-11:
to add to that, 7.1.0 still comes with binding 2.0.5.

maxwellhadley, 2019-01-11:
7.1.0 still segfaults here, running macOS 10.13.6

reconbot, 2019-01-12:
I am happy to say `serialport@7.1.3` has been released which fixes this issue.

closed 2019-01-12 by reconbot