serialport/node-serialport #1771: Using SerialPort ends in Segmentation fault (Speicherzugriffsfehler)

Maintraxx, 2019-01-09:
### Summary of Problem

- What are you trying to do?
Using serialport for access the serial port.

- What happens?
A simple test program ends up in an error message Speicherzugriffsfehler ("segmentation fault"). 
Other ports (e.g. ttyAMA0) produce the same error.

- What should have happened?
No error message / crash should occure.

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')
const port = new SerialPort('/dev/ttyUSB0', { baudRate: 256000 })
```

### Versions, Operating System and Hardware

- SerialPort@?7.1.2
- Node.js v8.15.0
- Linux HomeControl_Panel1 4.14.90-v7+ #1183 SMP Fri Dec 21 14:03:50 GMT 2018 armv7l GNU/Linux
- Hardware and chipset? All (internal UART and CH341)

Maintraxx, 2019-01-09:
In the meanwhile, pierreroth64  postet the same issue. I can confirm, what he descriped. When i change to 6.2.2 everythink works.

pierreroth64, 2019-01-09:
Here is mine #1770

nowakpl, 2019-01-10:
https://github.com/node-serialport/node-serialport/blob/8fc58a429bbc1483589897b9d3d497ffdaac16cf/packages/bindings/src/serialport_unix.cpp#L284

SIGSEGV happens at that line because ConnectionOptionsBaton inherits from Nan::AsyncResource (requiring a v8 isolate) and EIO_Open -> setup is called from a uv thread that doesn't have an associated isolate.

mmccool, 2019-01-10:
I had the same problem.  Tried different versions of Node (eg LTS, latest, 8.15, etc) with no change.

luccasmaso, 2019-01-10:
Same here, downgrading to 6.2.2 works

reconbot, 2019-01-10:
How does 7.1.0 work for you?

reconbot, 2019-01-10:
We should have this fixed later today btw #1772

reconbot, 2019-01-12:
[`serialport@7.1.3`](https://www.npmjs.com/package/serialport/v/7.1.3) has been relased with the fix. Sorry for the delay on this.

closed 2019-01-12 by reconbot