serialport/node-serialport #1773: Reading incoming data with "inter_byte_timeout" like in pySerial?

AfdhalAtiffTan, 2019-01-10:
Hi,

I am very new to javascript, I've been using pySerial's "inter_byte_timeout" feature that allows instead of using '\n' as the delimiter, it uses time-based "delimiter" that will separate previous incoming packet from a new one.

Basically, I have a streaming hex-data (not ascii, so delimiter is not usable here) that are separated by 15ms apart.

Is there any way to achieve this?

Many thanks.

HipsterBrown, 2019-01-10:
Hello @AfdhalAtiffTan 

`node-serialport` does not have a time-based parser at the moment, but you could create your own. https://serialport.io/docs/en/api-parsers-overview

If you learn about node.js transform streams and emit data from the stream every 15ms.

Holger-Will, 2019-01-11:
the parser could be implemented like this:

```
const Transform = require('stream').Transform

class MyParser extends Transform {
  constructor (options = {interval: 15}) {
    super()
    this.currentPacket = []
    setInterval(this.emitPacket.bind(this), options.interval)
  }
  _transform (chunk, encoding, cb) {
    for (var offset = 0; offset < chunk.length; offset++) {
      this.currentPacket.push(chunk[offset])
    }
    cb()
  }
  emitPacket () {
    this.push(Buffer.from(this.currentPacket))
    this.currentPacket = []
  }
}
```

Holger-Will, 2019-01-11:
and you would use it like this:

```
const SerialPort = require('serialport')
const port = new SerialPort('/dev/ttyACM0', { baudRate: 9600 })
const parser = port.pipe(new MyParser({interval:1000})) //emit packet every 1s
parser.on('data', console.log)
```

reconbot, 2019-01-12:
maybe emit data only if there is data to emit? It's a good parser, I'd take a pr for it

Holger-Will, 2019-01-12:
after reading up a bit on `inter_byte_timeout` i realize it behaves a bit differently. i would implement the parser so it matches that behavior. 

> inter_byte_timeout: Starts the count down clock over, every time a byte is received. If the characters come in continuous stream with fx 1ms between bytes, it can go on forever if the inter_byte_timeout is just greater than 1ms.

what i get from this is, that the parser should only emit a packet, if there is no data for `interval` ms (It does what it says inter-byte-timeout... ;-) )
packets are separated by intervals of silence.

i would call it `InterByteTimeout`-Parser to give a hint that it matches pySerials behavior...

Holger-Will, 2019-01-12:
there is a second timeout setting, defined like this:

> timeout = x: set timeout to x seconds (float allowed) returns immediately when the requested number of bytes are available, otherwise wait until the timeout expires and return all bytes that were received until then.

would it makes sense to extend the byte-length-parser to have a timeout setting?

AfdhalAtiffTan, 2019-01-12:
@HipsterBrown Thank you for the pointer to Stream, I am learning about it now.
@Holger-Will Thank you for the parsers.

I just tested the following on my Arduino Uno to simulate a streaming packet spaced at 5ms apart (on my oscilloscope, it is closer to 10ms):
`
void setup() {
  Serial.begin(115200);
}

void loop() {  
  static unsigned long prev_millis = 0;
  if(millis() - prev_millis > 5) {
    prev_millis = millis();
    Serial.print("Elapsed time: ");
    Serial.print(millis()); //no '\n' delimiter
  }
}
`

My javascript is as follows:
'
const SerialPort = require('serialport')
const port = new SerialPort('COM8', { baudRate: 115200 })
port.on('data', logger)

function logger(data) {
	console.log('<start> ' + data.toString('hex') + ' <end>');
}
'

Surprisingly, it is able to separate incoming packets pretty fast, maybe even faster than pySerial. 
But, it would be nice if there is a way to control the inter_char_timeout. I think I am happy with the performance for now.

With your 'myParser', it tends to split and merge the packets at a random byte. I am dealing with unpredictable asynchronous data, so, I can understand why it is non-trivial. The gap between messages can vary between 5ms to 2s. 15ms is the average.

'InterByteTimeoutParser' doesn't print anything, I believe this is because of my limited experience with javascript.

I had issues with pySerial before because it cannot separate incoming packets fast enough and I ended up having to deal with merged bytes and separate it manually. 

Overall, I think I am ok for now. As long as serialport can discriminate between new packet from the old one, it will work with my project.

Thank you all again!

Holger-Will, 2019-01-21:
@reconbot thanks for merging. can this be closed now?

reconbot, 2019-01-26:
I think so =)

reconbot, 2019-01-26:
Published as ‘@serialport/parser-inter-byte-timeout’

closed 2019-01-26 by reconbot