serialport/node-serialport #1775: Ingenico protocol

bgnq, 2019-01-11:
Hello,

I can not implement a protocol to control a payment terminal in windows OS with the last version on serialport.
I attach specification and my code.
Connexion is OK but the write function do nothing.

Can you help me please.
Thank a lot !

My code : 
https://gist.github.com/bgnq/d0be0ce88c4d34cb7f940319da01fb06

Render with debug  : 
http://www.zupimages.net/viewer.php?id=19/02/mtf5.png

Protocol explication : 
http://www.zupimages.net/viewer.php?id=19/02/k9tg.png
http://www.zupimages.net/viewer.php?id=19/02/jwm4.png

JBtje, 2019-01-11:
```js
myPort.write('ENQ');
```

You are writing the string ENQ to the device... but your protocol tells you to write the character with the name `ENQ`, which is written as 0x05.

https://en.wikipedia.org/wiki/ASCII

Just like "A" is the 'name' of the character represented by 0x41

tip: `myPort.write( 0x05 );` I assume that will work...

bgnq, 2019-01-11:
Thanks!
If I try :
myPort.write( 0x05 );
I get error : string or buffer type for parameter of function write

Thank for your help

Holger-Will, 2019-01-11:
`myPort.write(Buffer.from([0x05]);`

bgnq, 2019-01-14:
Yes !
Thanks your everybody.

closed 2019-01-14 by bgnq