serialport/node-serialport #1781: While using iterator the write callback will be very slow

pucumt, 2019-01-14:
I am using iterator to make node sync print, but then the write callback will be very slow after sometimes.
You may need call iterator3 more than 3 times, then it will be very slow.
### Code to Reproduce the Issue
```js
function iterator3(curIndex3) {
                    if (curIndex3 >= maxlength) {
                        return true;
                    }
                    return labelPrint()
                        .then(() => {
                            return iterator3(curIndex3 + 1);
                        });
                };
                return iterator3(0);
function labelPrint() {
        return new Promise(function (resolve, reject) {
             labelPort.write('PRINT 1\r\n', function () {
                 resolve(true);  // this is the place very slow after more than 3 times
             });
        });
    };
```
while debug, the issue might be in promisify of util.js
But I don't know how to change it.

The process to `resolve(data);` is very quick, but the promise back to `this.writeOperation = null` is very very slow.

### Versions, Operating System and Hardware
- SerialPort@@serialport/bindings
- Node.js v10.15
- Windows10

HipsterBrown, 2019-01-14:
@pucumt Can you share your debug logs?

My initial thought is it could be a threading issue on the windows c++ bindings based on the source comment:

```
// WriteFileEx requires a thread that can block
```

Each thread should be cleared once the write operation resolves. What are you writing to? Could that be causing a bottleneck after a few writes?

Also, to help with the readability of the code, you could try using [async iterators](https://www.codementor.io/tiagolopesferreira/asynchronous-iterators-in-javascript-jl1yg8la1):

```es6
function labelPrint() {
        return new Promise(function (resolve, reject) {
             labelPort.write('PRINT 1\r\n', function () {
                 resolve(true);  // this is the place very slow after more than 3 times
             });
        });
    };
const asyncIterable = {
  [Symbol.asyncIterator]: async function* asyncGenerator() {
    let currentIndex = 0;
    while (currentIndex < maxLength) {
      // it waits for the promise to resolve
      // before yield the value
      currentIndex += 1;
      yield await labelPrint();
    }
  }
};
// "top-level" await
(async () => {
  for await (const item of asyncIterable) {
    console.log('labelPrint has resolved');
  }
})();
```

pucumt, 2019-01-15:
@HipsterBrown I am using vs code to debug node, so there is no log.
And the async await function has the same issue. 
I agree with you that the thread might be block, but I cannot find a way to avoid it.

Hope there will be a solution in the no far future.

Thank you very much!

HipsterBrown, 2019-01-15:
@pucumt You can get debug logs from serialport when running the program by settings the `DEBUG` environment variable -> https://serialport.io/docs/en/guide-debugging#windows-command-prompt-notes

pucumt, 2019-01-18:
[image]

HipsterBrown, 2019-01-18:
@pucumt What port and type of device are you writing to?

closed 2019-03-26 by stale[bot]