serialport/node-serialport #1782: when  coming large amounts of  binary data , receive some error codes

hustzju, 2019-01-16:
My embedded device sends large amounts of  binary data(more than 700k),  the app developed by electron with node-serialport  receives the data than handle it. But I find  the received data always  make mistakes, sometimes Receive a few more bytes, sometimes Receive a few less bytes, sometimes Receive a few error bytes.

when I run the app on win7, the issue happens, but win10 is OK.
I also develop another app  by windows MFC using other C++ serialport library, and  receive data from the same source, everything is OK.

SO I wonder if node-serialport has a compatibility problem  with win7 when receiving large amounts of  binary data

neither of  the two methods of receiving data  blow works.
// Code

//Method 1
   serialPort.on('readable', () => {
        let chunk;
        while((chunk = serialPort.read()) !== null){
        length += chunk.length;
        BufferData.push(chunk);
    
        }
    });
	
//Method 2
    serialPort.on('data', function (data) {
        length += data.length;
        BufferData.push(data);
       
    });

### Versions, Operating System and Hardware

- SerialPort@?
serialport: 7.1.1
- Node.js v?
node 8.2.1  x86
- Windows? Linux? Mac?
win7

HipsterBrown, 2019-01-16:
@hustzju Thanks for creating an issue. Can you share the debug output from your program on window 7? https://serialport.io/docs/en/guide-debugging#cmd

hustzju, 2019-01-22:
> @hustzju Thanks for creating an issue. Can you share the debug output from your program on window 7? https://serialport.io/docs/en/guide-debugging#cmd

thanks for your reply,  and sorry for  replying to you later. I have got the debug output from the bug app on window 7 . it just like this:
browser.js:183 serialport/stream binding.read +201ms finished
browser.js:183 serialport/stream _read +3ms reading
src\browser.js:183 serialport/binding-abstract read +233ms
browser.js:183 serialport/stream binding.read +6ms finished
browser.js:183 serialport/stream _read +2ms reading
src\browser.js:183 serialport/binding-abstract read +8ms
.......
except  the three same lines above , I also Capture 9 same lines like that:
browser.js:183 serialport/stream _read +1ms discarding the read buffer pool

Beyond these  lines,   no  other different messages
The entire log file is so big, and I attach it  int the attachment.
[Uploading win7_receive.log…]()

I really appreciate your next  anwser.

hustzju, 2019-01-29:
I have uploaded the debug output, could you give me some advice?

closed 2019-04-06 by stale[bot]