serialport/node-serialport #1785: Latest prebuild binaries missing (@serialport/bindings@2.0.6)?

bkniffler, 2019-01-16:
Hey, 

Since I've been running into problems and I noticed https://github.com/node-serialport/node-serialport/releases does not list any binaries under latest @serialport/bindings@2.0.6 tag, are the current binaries missing?

HipsterBrown, 2019-01-16:
@bkniffler Thanks for creating an issue. I'll look into our CI process and see what's up.

bkniffler, 2019-01-16:
ps. The error won't go away unfortunately if I downgrade to some other version. Trying to build an electron app targeting windows from my mac dev machine. This should be working though thanks to prebuild right? If yes, I'd just wait for latest binaries to be released and retry.

HipsterBrown, 2019-01-16:
What are your node and electron versions? 

Ci is building the latest binaries now, which should take ~30 minutes.

bkniffler, 2019-01-16:
Got node v10.14.2 locally and tried electron@3.1.1 and electron@4.0.1.

HipsterBrown, 2019-01-17:
@bkniffler The prebuilt binaries for `@serialport@bindings@2.0.6` are available now -> https://github.com/node-serialport/node-serialport/releases/tag/%40serialport%2Fbindings%402.0.6

HipsterBrown, 2019-01-17:
@reconbot Yes, I repushed it.

bkniffler, 2019-01-17:
All works nice now :) Thanks