serialport/node-serialport #1787: Can not write more than one character if getting response from the device for the character

aarshkd, 2019-01-23:
### Summary of Problem
- What are you trying to do?
I trying to make a application for the device I am working on. The device sends various data based on characters it receives. As an example, it sends the output of the sensor if it receives 'A' and it outputs current version of a firmware if it receives 'V'.

- What happens?
I have setup on 'data' event that listens to the data it receives. I have put setInterval(function, 1000) that sends 'A' character(port.write('A')) to the device and device sends the output back. It is works properly. But If I try to send characters 'A' and 'V' at a same time(port.write('AV')), the device doesn't receive character 'V' as device is sending data back for the character 'A'. If I try sending 'T'(Device sends nothing back for 'T') and 'V' it receives character 'V' and sends the current version correctly.

### Code to Reproduce the Issue
    function openPort() 
    {
      //Opening port manually
         port.open(function (err) {
          if (err) {
            console.log(err)
          }
     });

    const Readline = serialport.parsers.Readline;	
    const parser = new Readline();			
    port.pipe(parser);
    parser.on('data', onData);
    intervalID = setInterval(Apoll, 1000);
    };

    function onData(data){
      console.log('data:' + data);
    }

    function Apoll(){
       port.write("A");
       //port.write("AV"); // will ignore 'V'
       //port.write("TV"); // Will send both character
    }

### Versions, Operating System and Hardware

- SerialPort@ 7.1.3
- Node.js v 10.15
- Windows 10
- Hardware and chipset? (Prolific/FTDI/Other)  - PIC Controller

fivdi, 2019-02-02:
> But If I try to send characters 'A' and 'V' at a same time(port.write('AV')), the device doesn't receive character 'V' as the device is sending data back for the character 'A'.

Are you 100% sure that the device doesn't receive the character 'V'? Or are you assuming that the device doesn't receive the character 'V' because it doesn't send the response for 'V'?

I ask these questions because perhaps the device doesn't handle valid commands correctly if they are sent to the device too quickly.

fivdi, 2019-02-03:
I just tried something similar (without electron) to see if it was possible to find an issue with serialport.

With the following firmware on an Arduino UNO:
```c++
void setup() {
  Serial.begin(9600);
}

void loop() {
  if (Serial.available()) {
    char ch = char(Serial.read());
    if (ch == 'A') {
      Serial.println("Sensor data: 1234");
    } else if (ch == 'V') {
      Serial.println("Version: v1.0.0");      
    }
  }
}
```
And the following program using serialport on a PC:
```js
const SerialPort = require('serialport');
const parser = new SerialPort.parsers.Readline;
const port = new SerialPort('/dev/ttyACM0', {
  baudRate: 9600
});

port.pipe(parser);

parser.on('error', (err) => {
  console.log('Error: ', err.message);
});

parser.on('data', (data) => {
  console.log(data);
});

port.on('open', () => {
  setInterval(() => {
    port.write('AV');
  }, 1000);
});
```
The output to the console on the PC was:
```
Sensor data: 1234
Version: v1.0.0
Sensor data: 1234
Version: v1.0.0
Sensor data: 1234
Version: v1.0.0
Sensor data: 1234
Version: v1.0.0
Sensor data: 1234
Version: v1.0.0
Sensor data: 1234
Version: v1.0.0
Sensor data: 1234
Version: v1.0.0
Sensor data: 1234
Version: v1.0.0
...
```
Everything works as expected. I'm afraid I'm not familiar with electron and can't provide any information related to electron.

> I have tried debugging the firmware. It never stops at the breakpoint in UARTisr in the case I stated above.

Debugging interrupt routines can be extremely difficult. For example, if you set a breakpoint at the start of the interrupt routine and saw that the character 'A' was received and the serialport program keeps sending data in the background there will very likely be overrun errors on the device. Is it possible that there are overrun errors on the device? If so, are they handled correctly? How many characters can the hardware UART on the device receive and buffer before there are overrun errors?

How are the responses sent from the firmware on the device to the serialport program on the PC? Are they sent synchronously by the interrupt routine? If so, this will result in "huge" interrupt handling delays further increasing the possibility of overrun errors.

> I am just a beginer for electron and node js. Is there any suggestion that I can test or try?

The program posted in the initial post above is incomplete, has syntax errors and therefore doesn't work :smile: This makes it difficult to determine if there is something wrong with the program. One thing is missing in the program, it doesn't wait for an 'open' event on the port before sending data to the device. This means that the program will start sending data to the device in Apoll after a 1 second delay irrespective of whether the port has been successfully opened or not.

I think that serialport is functioning correctly and sending all the required data to the device.

aarshkd, 2019-02-04:
@fivdi Thank you very much for your time and small test case. You are right. I have taken over the code that is written by someone who doesn't have any programming knowledge. Everything is happening inside the UART routine in firmware. There are lot of unnecessary lines inside the ISR and that is causing issue. I have to re write half of the code again so that I don't face these issues in the future. 

The node js code I posted above contains only issue specific part. Connection is established by the user through button. The open event is there in the remaining code. I simply forgot to post it here.  

Again thanks for your time and help. Serialport is working correctly. I will close the issue if I have a privilege.

closed 2019-02-04 by aarshkd