serialport/node-serialport #1791: serialport makes devtools disconnected from the page.

wiener0zyj, 2019-01-28:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
I'm tring to run the demo electron-serialport on win10. after i build and start up the electron. 
I get this this error: "devtools disconnected from the page" and nothing showing on the screen.
enviroment:
win10
electron 4.4.0
serialport  7.1.4

- What happens?
"devtools disconnected from the page"

- What should have happened?
I think the electron crashes! I could not decide what to do.

### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)
serialport.list(function(err, ports) {
        console.log(ports);
        });
// Code
```

### Versions, Operating System and Hardware

- SerialPort@? 7.1.4
- Node.js v? 10.15.0  x64
- Windows? Linux? Mac? Windows 10
- Hardware and chipset? (Prolific/FTDI/Other) did not connect to hardware

wiener0zyj, 2019-01-28:
This blocks me for two days. I have rebuild the electron. but what ever codes about serialport  added would crashed electron. Too sad.
I have even tried to build from source code, still problems.
anyone has encountered the same problem? please give me a way.

peteshand, 2019-01-28:
I'm having the same issue.
Electron 4.0.2
Node 11.11.0
Mac 10.14

I've rebuild from source. As soon as I call the serialport constructor with a path devtools disconnects

aarshkd, 2019-01-29:
I was facing this issue before. I couldn't find the exact issue causing this but deleting everything related to electron(Locally and Globally) solved it. Try making a fresh install.

peteshand, 2019-01-29:
similar story to me.. I got it working yesterday as well by deleting everything and then got it working with https://github.com/helloyan/electron-serialport and then transitioned over to this repo

closed 2019-04-06 by stale[bot]