serialport/node-serialport #1794: serialport.list callback fails on macOS if called more than once

SeBassTian23, 2019-02-04:
Hi,

I recently encountered the following issue using serialport for an Electron app.

The app was compiled under macOS 10.13.x and the function `serialport.list` was returning a list of ports every time it was called. This is the case for macOS 10.13x and 10.14.x.

After an update the app was compiled using macOS 10.14.2. On macOS 10.14.x everything works as expected, but under 10.13.x the `serialport.list` function returns the list of ports only once. After a second time the callback never returns, so there is no error message or anything. Using the console I tired it manually inside the app and I get the same result. So far the app needs to be restarted to list the ports again.

However, when compiling the updated app on macOS 10.13.x it works on both versions as expected.

I tested the release 6.2.2 as well as 7.1.4 with node v8.11.2 and v10.15.1, all giving me the same result.

Any ideas on what might be going on? Please let me know if I can provide further information.

```js
const SerialPort = new SerialPort(path)

// Code
function SerialGetAllPorts(selected) {

    SerialPort.list(function (error, ports) {
        if (error)
            throw error;

        ports.forEach(function (port) {
            // Excluding ports not used for connections
            if( port.comName.match(/Bluetooth-Incoming-Port/ig) )
                return;

            var name = port.comName.replace(/(\/dev\/(cu|tty)?\.?)/g, '');
            if (port.manufacturer !== undefined)
                name = port.manufacturer + ' (' + name + ')';
            console.log( port.comName, name);
        });
        ...
    });
}
```

- SerialPort@6.2.2 & SerialPort@7.1.4
- Node.js v8.11.2 or v10.15.1
- Mac

rodneym, 2019-02-10:
Experiencing the same behavior..

SerialPort@7.1.1
Node.js V10.15.0
MacOS 10.13.6

SeBassTian23, 2019-03-18:
Any leads or insights into this issue?
I'm happy to help and test!

gniezen, 2019-03-22:
This is related to a similar issue in libusb (https://github.com/libusb/libusb/issues/519) where when it's built using Mojave, it's not backwards-compatible with High Sierra.

TL;DR: IOUSBLib.h is broken and Apple would need to update XCode to fix it. In the meanwhile, a similar fix like https://github.com/libusb/libusb/commit/efa1b280e66c0a62d892bddf826e624362d1e50a could be used?

The affected code in node-serialport is here:
https://github.com/node-serialport/node-serialport/blob/master/packages/bindings/src/darwin_list.cpp#L245

reconbot, 2019-04-04:
@gniezen that looks promising

SeBassTian23, 2019-04-04:
It seems like the latest update of macOS Mojave 10.14.4 and the included update of xcode command line tools fixed the issue.

xcode-select version 2354

reconbot, 2019-04-08:
nice!

closed 2019-04-08 by reconbot