serialport/node-serialport #1797: Exception throw during inspection when using Node 11.x

jacobq, 2019-02-07:
### Summary of Problem

(Please answer all 3)

- What are you trying to do? 

Peek at the `port` object (returned by `new SerialPort(...)`) in the node 11.x (11.9.0) REPL.

- What happens?

This causes an unhandled exception to be thrown up:
```
RangeError: Invalid count value
    at String.repeat (<anonymous>)
    at reduceToSingleString (internal/util/inspect.js:1216:27)
    at formatRaw (internal/util/inspect.js:777:15)
    at formatValue (internal/util/inspect.js:539:10)
    at inspect (internal/util/inspect.js:196:10)
    at Object.formatWithOptions (util.js:174:18)
    at Console.(anonymous function) (internal/console/constructor.js:274:15)
    at Console.log (internal/console/constructor.js:283:59)
```

- What should have happened?

On Node 10.x (tested 10.15.1) no exception is thrown, and some interesting information about the object is returned:

<details>
<summary>Expected output</summary>

```
SerialPort {
  _readableState:
   ReadableState {
     objectMode: false,
     highWaterMark: 65536,
     buffer: BufferList { head: null, tail: null, length: 0 },
     length: 0,
     pipes:
      ReadLineParser {
        _readableState: [ReadableState],
        readable: true,
        _events: [Object],
        _eventsCount: 7,
        _maxListeners: undefined,
        _writableState: [WritableState],
        writable: true,
        allowHalfOpen: true,
        _transformState: [Object],
        includeDelimiter: false,
        delimiter: <Buffer 0d>,
        buffer: <Buffer > },
     pipesCount: 1,
     flowing: true,
     ended: false,
     endEmitted: false,
     reading: true,
     sync: false,
     needReadable: true,
     emittedReadable: false,
     readableListening: false,
     resumeScheduled: false,
     emitClose: true,
     destroyed: false,
     defaultEncoding: 'utf8',
     awaitDrain: 0,
     readingMore: false,
     decoder: null,
     encoding: null },
  readable: true,
  _events:
   [Object: null prototype] {
     end:
      { [Function: bound onceWrapper] listener: [Function: onend] },
     data: [Function: ondata] },
  _eventsCount: 2,
  _maxListeners: undefined,
  _writableState:
   WritableState {
     objectMode: false,
     highWaterMark: 65536,
     finalCalled: false,
     needDrain: false,
     ending: false,
     ended: false,
     finished: false,
     destroyed: false,
     decodeStrings: true,
     defaultEncoding: 'utf8',
     length: 0,
     writing: false,
     corked: 0,
     sync: false,
     bufferProcessing: false,
     onwrite: [Function: bound onwrite],
     writecb: null,
     writelen: 0,
     bufferedRequest: null,
     lastBufferedRequest: null,
     pendingcb: 0,
     prefinished: false,
     errorEmitted: false,
     emitClose: true,
     bufferedRequestCount: 0,
     corkedRequestsFree:
      { next: null,
        entry: null,
        finish: [Function: bound onCorkedFinish] } },
  writable: true,
  allowHalfOpen: true,
  binding:
   LinuxBinding {
     bindingOptions: { vmin: 1, vtime: 0 },
     fd: 21,
     writeOperation: null,
     openOptions:
      { vmin: 1,
        vtime: 0,
        autoOpen: true,
        baudRate: 115200,
        dataBits: 8,
        hupcl: true,
        lock: true,
        parity: 'none',
        rtscts: false,
        stopBits: 1,
        xany: false,
        xoff: false,
        xon: false,
        highWaterMark: 65536 },
     poller:
      Poller {
        _events: [Object],
        _eventsCount: 1,
        _maxListeners: undefined,
        poller: Poller {} } },
  path: '/dev/ttyUSB0',
  settings:
   { autoOpen: true,
     baudRate: 115200,
     dataBits: 8,
     hupcl: true,
     lock: true,
     parity: 'none',
     rtscts: false,
     stopBits: 1,
     xany: false,
     xoff: false,
     xon: false,
     highWaterMark: 65536 },
  opening: false,
  closing: false,
  _pool:
   <Buffer 30 30 35 30 0d 30 2e 30 30 45 2b 30 30 2c 30 30 35 30 2c 33 39 2e 30 2c 32 32 2e 30 2c 34 30 2e 30 2c 33 30 2e 39 2c 39 37 2e 35 2c 30 2e 30 2c 30 2e ... >,
  _kMinPoolSpace: 128 }

```

</details>

### Code to Reproduce the Issue

<details>
<summary>Minimal program</summary>

```js
const SerialPort = require('serialport');
const port = new SerialPort('/dev/null', { autoOpen: false });
console.log(port); // Throws
/*
RangeError: Invalid count value
    at String.repeat (<anonymous>)
    at reduceToSingleString (internal/util/inspect.js:1216:27)
    at formatRaw (internal/util/inspect.js:777:15)
    at formatValue (internal/util/inspect.js:539:10)
    at Object.inspect (internal/util/inspect.js:196:10)
    at REPLServer.self.writer (repl.js:514:33)
    at finish (repl.js:686:38)
    at finishExecution (repl.js:310:7)
    at REPLServer.defaultEval (repl.js:396:7)
    at bound (domain.js:393:14)
*/
```

</details>

<details>
<summary>Longer program</summary>

```js
const SerialPort = require('serialport');
const Readline = SerialPort.parsers.Readline

console.log('test program started');
SerialPort.list().then((ports) => {
    console.log('list() resolved -->', ports);
    // Open first port we see
    const comName = ports[0].comName;
    const openOptions = {
        baudRate: 115200,
    };
    console.log(`About to open/create port using "${comName}"`, openOptions);
    const port = new SerialPort(comName, openOptions);
    const parser = new Readline({ delimiter: '\r', });
    console.log('Attaching parser');
    port.pipe(parser);
    console.log('Attaching receive event handler');
    parser.on('data', console.log.bind(console));

    port.write('command 1\r\n');

    new Promise(resolve => { setTimeout(resolve, 1000); }).then(() => port.write('command 2\r\n'));

    new Promise(resolve => { setTimeout(resolve, 2000); }).then(() => {
        console.log('Attempting to inspect "port" object throws error', port); // RangeError: Invalid count value
    });
});
console.log('waiting for .list() promise to resolve');
```

</details>
### Versions, Operating System and Hardware

- SerialPort v7.1.4
- Node.js v11.9.0 (works on 10.15.1)
- Linux (4.9.0-8-amd64, Debian)
- Hardware and chipset? FTDI but doesn't appear to matter (can reproduce with others too)

jacobq, 2019-04-08:
Just tested again and can confirm that this problem is still happening on node 11.13.0 and Linux x86_64 (Debian / kernel 4.9.0-8-amd64). Is no one else encountering this / able to reproduce?

vladh, 2019-05-05:
I am also encountering this on Node v11.14.0 on macOS 10.14.4.

jacobq, 2019-05-05:
Just tested again (`serialport` v7.1.5) on Windows with Node v11.15.0 and v12.1.0. It looks like the problem happens for 11 but not 12. (I am no longer using Node 11 so am not too interested in investigating this further.)

<details>
<summary>Node 11</summary>

```
npm install serialport

> @serialport/bindings@2.0.8 install C:\Users\jacob\Documents\tmp\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

npm WARN saveError ENOENT: no such file or directory, open 'C:\Users\jacob\Documents\tmp\package.json'
npm WARN enoent ENOENT: no such file or directory, open 'C:\Users\jacob\Documents\tmp\package.json'
npm WARN tmp No description
npm WARN tmp No repository field.
npm WARN tmp No README data
npm WARN tmp No license field.

+ serialport@7.1.5
added 78 packages from 45 contributors and audited 143 packages in 4.086s
found 0 vulnerabilities

C:\Users\jacob\Documents\tmp>node
> const SerialPort = require('serialport');
undefined
> const port = new SerialPort('/dev/null', { autoOpen: false });
undefined
> console.log(port); // Throws
Thrown:
RangeError: Invalid count value
    at String.repeat (<anonymous>)
    at reduceToSingleString (internal/util/inspect.js:1365:27)
    at formatRaw (internal/util/inspect.js:791:15)
    at formatValue (internal/util/inspect.js:529:10)
    at inspect (internal/util/inspect.js:187:10)
    at formatWithOptions (internal/util/inspect.js:1408:12)
    at Console.(anonymous function) (internal/console/constructor.js:277:10)
    at Console.log (internal/console/constructor.js:287:61)
> process.version
'v11.15.0'
```

</details>

<details>
<summary>Node 12</summary>

```
C:\Users\jacob\Documents\tmp>npm install serialport
...
+ serialport@7.1.5
added 78 packages from 45 contributors and audited 143 packages in 3.417s
found 0 vulnerabilities

C:\Users\jacob\Documents\tmp>node
Welcome to Node.js v12.1.0.
Type ".help" for more information.
> const SerialPort = require('serialport');
undefined
> const port = new SerialPort('/dev/null', { autoOpen: false });
undefined
> console.log(port); // Throws
SerialPort {
  _readableState: ReadableState {
    objectMode: false,
    highWaterMark: 65536,
    buffer: BufferList { head: null, tail: null, length: 0 },
    length: 0,
    pipes: null,
    pipesCount: 0,
    flowing: null,
    ended: false,
    endEmitted: false,
    reading: false,
    sync: true,
    needReadable: false,
    emittedReadable: false,
    readableListening: false,
    resumeScheduled: false,
    paused: true,
    emitClose: true,
    autoDestroy: false,
    destroyed: false,
    defaultEncoding: 'utf8',
    awaitDrain: 0,
    readingMore: false,
    decoder: null,
    encoding: null
  },
  readable: true,
  _events: [Object: null prototype] {},
  _eventsCount: 0,
  _maxListeners: undefined,
  _writableState: WritableState {
    objectMode: false,
    highWaterMark: 65536,
    finalCalled: false,
    needDrain: false,
    ending: false,
    ended: false,
    finished: false,
    destroyed: false,
    decodeStrings: true,
    defaultEncoding: 'utf8',
    length: 0,
    writing: false,
    corked: 0,
    sync: true,
    bufferProcessing: false,
    onwrite: [Function: bound onwrite],
    writecb: null,
    writelen: 0,
    bufferedRequest: null,
    lastBufferedRequest: null,
    pendingcb: 0,
    prefinished: false,
    errorEmitted: false,
    emitClose: true,
    autoDestroy: false,
    bufferedRequestCount: 0,
    corkedRequestsFree: {
      next: null,
      entry: null,
      finish: [Function: bound onCorkedFinish]
    }
  },
  writable: true,
  allowHalfOpen: true,
  binding: WindowsBinding { bindingOptions: {}, fd: null, writeOperation: null },
  path: '/dev/null',
  settings: {
    autoOpen: false,
    baudRate: 9600,
    dataBits: 8,
    hupcl: true,
    lock: true,
    parity: 'none',
    rtscts: false,
    stopBits: 1,
    xany: false,
    xoff: false,
    xon: false,
    highWaterMark: 65536
  },
  opening: false,
  closing: false,
  _pool: <Buffer 80 b5 37 46 5c 02 00 00 f0 c7 3a 46 5c 02 00 00 60 40 00 00 00 00 00 00 90 02 3d 46 5c 02 00 00 01 0e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ... 65486 more bytes, used: 0>,
  _kMinPoolSpace: 128
}
undefined
```

</details>

reconbot, 2019-05-05:
Since node 11 isn't LTS I'm not super interested in supporting it. However I would land any patches that make it work.

closed 2019-07-11 by stale[bot]