serialport/node-serialport #1800: Serial port data shows numbers even with readline parser

ShocKwav3, 2019-02-10:
### Summary of Problem
Reading from serial is not showing my data. It shows some numbers.

- What are you trying to do?
Sending string from arduino through serial port.

- What happens?
I see numbers in my node js console

- What should have happened?
I should have seen the strings that are sent from arduino

### Code to Reproduce the Issue
Arduno (Arduino mega):
```
int pushButton = 7;

void setup() {
  Serial.begin(9600);
  pinMode(pushButton, INPUT);
}

void loop() {
  int buttonState = digitalRead(pushButton);

  if(buttonState == HIGH){
    Serial.println('ononon');
    digitalWrite(LED_BUILTIN, LOW);
  }else{
    Serial.println('offfff');
    digitalWrite(LED_BUILTIN, HIGH);
  }

  delay(100);
}
```

Node.js code (Raspberry pi 3 B+):
```
const five = require("johnny-five")
const Raspi = require("raspi-io")
const SerialPort = require("serialport")
const Readline = require('@serialport/parser-readline')

const arduinoSerialPort = new SerialPort("/dev/ttyACM0", {
  baudRate: 9600
})
const parser = arduinoSerialPort.pipe(new Readline({ delimiter: '\r\n' }))

const board = new five.Board({
  io: new Raspi(),
  repl: false,
})

board.on('ready', function() {
  parser.on('data', console.log)
})

```
### Versions, Operating System and Hardware

- SerialPort: 7.1.4
- Node.js: v11.9.0
- Linux (Raspbian Stretch)
- Hardware and chipset: Raspberry Pi 3 B+

gganebnyi, 2019-02-13:
What is the output?

ShocKwav3, 2019-02-14:
Actually I messed it up on the arduino code. I used single quote for strings (too much javascripting :p)
But I got it fixed now. Thanks

closed 2019-02-14 by ShocKwav3