serialport/node-serialport #1802: A dynamic link library (DLL) initialization routine failed

amapleaf, 2019-02-14:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
I want to develop a VS code extension that includes serial port operations.
- What happens?
The VS code was updated to 1.31 in the past two days. I found that my extension could not be used. When debugging, I got a DLL initialization failure error.
- What should have happened?
After I downloaded my extension, I executed `npm install` and then executed `npm rebuild`(this is because there is a problem with `NODE_MODULE_VERSION` mismatch). After pressing F5, the extension cannot be started.

### Code to Reproduce the Issue

```ts
import * as SerialPort from 'serialport';
import * as vscode from 'vscode';

export function activate(context) {
    let disposable = vscode.commands.registerCommand('extension.helloworld', () => {
        vscode.windows.showInformationMessage('Hello, world');
        let port = new serialport('COM4', {
            baudRate: 115200,
            autoOpen: true
        });
        port.on('open', () => {
            vscode.window.showInformation('COM4 Open');
        });
    });
    context.subscriptions.push(disposable);
}
// Code
```

### Versions, Operating System and Hardware

- SerialPort@? 7.1.4 & 6.2.2
- Node.js v? 10.15.1
- Windows? Linux? Mac? Windows 10 v1709
- Hardware and chipset? (Prolific/FTDI/Other) I don't know

I tried electron 3.x and 4.x and they didn't work.

jan-j, 2019-02-14:
I have the same problem with Electron 3.0.10. Seems like a problem with prebuilt binaries, because it works fine when `node-gyp rebuild` is forced to run.

amapleaf, 2019-02-15:
@jan-j I tried to executed `node-gyp rebuild` in `node-modules/serialport` and it printed `gyp info ok` with some yellow warnings. But it still doesn't work.

jan-j, 2019-02-15:
> @jan-j I tried to executed `node-gyp rebuild` in `node-modules/serialport` and it printed `gyp info ok` with some yellow warnings. But it still doesn't work.

@amapleaf In case of Electron you need to execute it specifying target and dist url like that:
```
node-gyp rebuild --target=3.0.10 --arch=x64 --dist-url=https://atom.io/download/atom-shell
```
where target is Electron version that you're using.

Sorry for not mentioning that before, I didn't notice that you're also trying to run it on Electron.

amapleaf, 2019-02-15:
@jan-j Thank you for your solution. I tried the method you mentioned, it might be useful for electron, but it doesn't work in VS code.

In addition, I stumbled upon the solution by adding a hook below the `scripts` in the `package.json` file in the VS code extension project:

```json
"install": "electron-rebuild -v 3.1.3",
```

3.1.3 here can be replaced with other 3.x versions.

closed 2019-04-23 by stale[bot]