serialport/node-serialport #1806: Flush not working

FliegenKLATSCH, 2019-02-17:
### Flush doesn't clear the buffer?

- port.pause(); port.flush(); port.resume();
- data event is emitted for every message received during the paused period / before the flush.

### Code to Reproduce the Issue

```js
const SerialPort = require("serialport");
const Readline = require('@serialport/parser-readline');

const port = new SerialPort("/dev/ttyUSB1", {
    baudRate: 9600, databits: 7, stopbits: 1, parity: 'even', lock: false
});
const parser = port.pipe(new Readline({delimiter: "!", encoding: 'ascii'}));

parser.on("data", function(data) {
    console.log(new Date() + ': ', data);
});

setTimeout(() => {
    console.log('pause...');
    port.pause();

    setTimeout(() => {
        console.log('flush...');
        port.flush(function(err) {
            console.log(err);
            console.log('resume...');
            port.resume();
        });
    }, 10000);

}, 10000);
```

### Versions, Operating System and Hardware

- SerialPort@7.1.4
- Node.js v10.13.0
- Raspberry Pi 3

thedayofcondor, 2019-05-09:
V7.1.5 I can confirm.

On Windows - flush callback is called but the buffers are never flushed.
On Linux - works maybe 30% of the times

As a test, I am connected to a serial device that sends a packet every second.

```javascript
const SerialPort = require('serialport')
const Readline = require('@serialport/parser-readline')

const port = new SerialPort(process.argv[2], {
    baudRate: 57600,
    autoOpen: true,
    rtscts: true
})

port.on('open', () => {
    console.log("port opened");
    port.flush((error) => {
        console.log("port flushed")       
        const parser = port.pipe(new Readline({ delimiter: '\r' }))
        parser.on('data', line => {
            console.log(line)
        })
    })
});
```
Upon launching (passing the serial port name as first parameter), the code above prints all the packed that have been received since the last launch notwithstanding the flush.

reconbot, 2019-05-09:
What behavior are you expecting?

reconbot, 2019-05-09:
Parsers and streams have their own buffers which are not flushed by this command. It only flushes the os level buffers.

thedayofcondor, 2019-05-09:
The behaviour I am expecting Is to only see the data sent after the flush.
Also, I am initialising the parser after the flush - so I do not expect it to be able to buffer anything.

But, if this is the case, should I flush the stream as well?

reconbot, 2019-05-09:
How do you know when the data was sent? Open => flush => read might not miss any data, no data is buffered when the port is closed.

thedayofcondor, 2019-05-09:
The device I am using outputs one line per second to the serial. It is an USB device based on a FTDI chip.

Just starting the above code should only print new data, one per second - it prints about a dozen of lines straight away. 

If I kill the process, wait 5 seconds, relaunch it - it prints 5 lines immediately, then one line every second.

If I understand what you are saying, the device itself, and not windows, could be the one doing the buffering - but then I would get exactly the same behaviour on Linux

reconbot, 2019-05-09:
Maybe, it might be paying attention to the line state when the port is closed and they’re might be different behavior between systems. Can you tell me more about the device?

thedayofcondor, 2019-05-09:
This is the device. I am implementing sl-can for a project
http://www.vscom.de/usb-can-plus-iso.html

thedayofcondor, 2019-05-13:
As an additional note, (may help with troubleshooting)

```javascript
port.on('open', () => {
  const old = port.read(65536);
  if (old!=null) {
    debug("Discarded", old.length, "bytes")
  }
});
```
doesn't work either - immediately after an open port.read returns null, and then my "data" event is flooded with stale packets. Which makes me think the port is not really "open" at that time.

thedayofcondor, 2019-05-15:
Hi, 
I had a chat with VSCOM support - they immediately pointed the finger at probable issues with HW handshaking - necessary because the VSCOM communicates over serial at 3Mbps.

Not sure how it plays out, but it looks like on startup port.flush does not see any data because there is no data to see (buffering is done by the device), so this is not an issue with port.flush in the first place

closed 2019-07-21 by stale[bot]