serialport/node-serialport #1808: How to install offline

AlexandreTortevois, 2019-02-20:
Hello,
I would like to know how install node-red-node-serialport in offline.
I try to copy the directory but i have an Illegal Instruction with an ia32 processor.
I launch command `npm install serialport --build-from-source` but nothing is do
How can i compile in "local mode" ?

reconbot, 2019-02-20:
Try npm rebuild

AlexandreTortevois, 2019-02-20:
it does not matter :(

AlexandreTortevois, 2019-02-20:
Witch is the command line to building again manually ?

To make this part : 
`> @serialport/bindings@2.0.7 install /usr/lib/node_modules/node-red-node-serialport/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

make: Entering directory '/usr/lib/node_modules/node-red-node-serialport/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Leaving directory '/usr/lib/node_modules/node-red-node-serialport/node_modules/@serialport/bindings/build'`

reconbot, 2019-02-20:
You can try `node-gyp rebuild` too but you might need `node-gyp` installed, I don't know how to use the built in one

closed 2019-04-28 by stale[bot]