serialport/node-serialport #1815: How to read Arduino mega 2560 sensor values ?

darkworks, 2019-03-05:
- What are you trying to do?

Am trying to ready DHT11 sensor value which is connected  to my Arduino Mega 2560 board. Am using Johnny five which lacks serial communication so i came across this project so i thought i could use it with johnny five.
am following this basic guide : https://www.instructables.com/id/Serial-Communications-with-Arduino/
so i wonder how can i get raw data from pin A0 as my sensor is connected to A0 pin.

DHT11 sensor : https://www.adafruit.com/product/386
DHT11 data sheet : https://akizukidenshi.com/download/ds/aosong/DHT11.pdf

- What happens?
here is my code but am not able to read pin A0

```
 var five = require("johnny-five");
const SerialPort = require('serialport');
var arduinoCOMPort = "COM4";
var Sensor_pin = "A0";

var arduinoSerialPort = new SerialPort(arduinoCOMPort, {  
 baudRate: 9600
});

arduinoSerialPort.on('open',function() {
  console.log('Serial Port ' + arduinoCOMPort + ' is opened.');
});

arduinoSerialPort.read();
```

- What should have happened?

should get raw data from pin A0

### Code to Reproduce the Issue
// added above

### Versions, Operating System and Hardware

- SerialPort@ latest just installed it 
- Node.js v 10
- Windows 7
- Hardware and chipset = Arduino Mega 2560 + DHT11 sensor

callen5914, 2019-03-06:
You’re so close!

Instead of saying:

`port.on(‘open’, function (){})`

Use: 

`port.on(‘data’, function(data){ console.log(data)})`

Sorry if the formatting of my response is bad. I’m typing from my phone. 

Cheers.

callen5914, 2019-03-06:
I think it’s also worth mentioning that you might not need to user your open function as the port will automatically be opened unless you use the auto open = false option.

callen5914, 2019-03-06:
Okay now that I am on a laptop:

Here's a better format to my answers above:
```js
var five = require("johnny-five");
const SerialPort = require('serialport');
var arduinoCOMPort = "COM4";
var Sensor_pin = "A0";

var arduinoSerialPort = new SerialPort(arduinoCOMPort, { baudRate: 9600 });

// Really don't need to use in my opinion
arduinoSerialPort.on('open',function() {
  console.log('Serial Port ' + arduinoCOMPort + ' is opened.');
});

// What works for me
arduinoSerialPort.on('data', function(data){
    console.log(data);
);
```
Now from my limited experience with this library, simple using that last `on('data')` function does nothing. It needs to be called somewhere. So here's how I've done it (May not be optimal).

```js 
listen = function() {
  const parser = arduinoSerialPort.pipe(new Readline({ delimiter: '\r\n' }))
  parser.on('data', (data) => { console.log('Message: ', data) })
}

listen()
```

Hope this help dude,

Cheers

darkworks, 2019-03-08:
thanks you sir :) ya right now am getting some raw data through

```
// What works for me
arduinoSerialPort.on('data', function(data){
    console.log(data);
);
```

however the listen function not worked.

mike-io, 2019-03-19:
What do you mean by the listen not working?

I quickly whipped up some demo node code here https://gist.github.com/mike-io/c23ae153a8085585adfa0689ee5ca9ff that goes nicely with some demo Arduino code here https://gist.github.com/mike-io/ffc3b9b81ddf74e49fd4413eb3da9f6f

In the demo I'm sending 3 sensor values as a string the format `"123,456,789\r\n"` (minus the quotes) from the Arduino, out over the serial port, into my PC.

The script uses the inbuilt node-serialport ReadLine parser, which strips off the `"\r\n"` newline bits and calls the parser's 'data' callback for every new line that arrives, meaning that the variable 'data' will now be `"123,456,789"` (again ignore the quotes). I split this into an array, with each item being separated by a comma. I can then get the values I want from the array and do something with them. I've chose 3 values, but you can send as many or as few as you like, just add/remove commas

Note that I have the line `arduinoParser.on('data', function(data) {....` instead of `arduinoSerialPort.on('open', function() {.....`. If you get the data callback on the port itself you'll end up with the raw data buffer, whereas the parser gives you a nice formatted string.

Using the two source files above, and an actual Arduino plugged into my Windows PC, gives the following output in my console:
```
A: 102, B: 105, C: 0
A: 102, B: 102, C: 0
A: 91, B: 85, C: 0
A: 66, B: 69, C: 255
A: 91, B: 89, C: 255
A: 92, B: 92, C: 255
A: 98, B: 92, C: 0
```

Hopefully that helps!

darkworks, 2019-04-05:
thank you very much :) i will play with it and will let u know how it goes it my end 

thanks

closed 2019-06-11 by stale[bot]