serialport/node-serialport #1816: Uncaught Error: A dynamic link library (DLL) initialization routine failed with Electron

gozzoo, 2019-03-12:
### Summary of Problem

- What are you trying to do?

I'm trying to use node-serialport with Electron

- What happens?

I get this exception in the console:
```
Uncaught Error: A dynamic link library (DLL) initialization routine failed.
\\?\C:\work\project\node_modules\@serialport\bindings\build\Release\bindings.node
    at process.module.(:4200/anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740)
    at Object.module.(:4200/anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620)
    at tryModuleLoad (internal/modules/cjs/loader.js:559)
    at Function.Module._load (internal/modules/cjs/loader.js:551)
    at Module.require (internal/modules/cjs/loader.js:658)
    at require (internal/modules/cjs/helpers.js:20)
    at bindings (C:\work\tansys\node_modules\bindings\bindings.js:81)
    at Object.<anonymous> (C:\work\tansys\node_modules\@serialport\bindings\lib\win32.js:1)

### Versions, Operating System and Hardware

- SerialPort@7.0.4
- Electron 3.1.6
- electron-rebuild: 1.8.4
- Windows 10 32bit

### Latest working version

- SerialPort@6.2.2

dan-sam, 2019-03-19:
What Node Version do you use?

gozzoo, 2019-03-19:
@dan-sam I don't think that it maters because I'm using serialport with electron, but just in case it does:

```
C:\work>node -v
v10.14.2
```

JamshedLatipov, 2019-03-19:
Hello there, I have the same problem. 
Node: 11.10.0
Electron: 4.10
electron-rebuild: 1.8.4
SerialPort@7.0.4 & 6.2.2

Please help me :)

dan-sam, 2019-03-21:
Could you guys try it with Electron 3.1.6?
You might have tried the following already, but just in case you didn't:

1) run: `npm uninstall --save serialport`
2) run: `npm install --save-dev electron@3.1.6`
3) Make sure you really installed electron-rebuild and you added it to your json file install hook: "install": "electron-rebuild". 
Under "scripts" you should have then:
```
"scripts": {
    "start": "electron .",
    "install": "electron-rebuild"
  }
```
4) run: `npm install `
5) Since serialport is not installed now (after step 1) the previous step (step 4) should compile and being rebuild by electron-rebuild
6) run: `npm install --save serialport`
7) run again: `npm install
`
I hope this will work

gozzoo, 2019-03-22:
@dan-sam thanks for your help.

What you are suggesting is exactly what I've been doing. Just to make sure I did it again, but unfortunatelly I'm getting the same result as before: 

`node_modules\bindings\bindings.js:88 Uncaught Error: A dynamic link library (DLL) initialization routine failed.`

closed 2019-05-28 by stale[bot]

Racafe92, 2019-09-10:
I had the same problem with electron v3.0.10 and the solution provided by @dan-sam worked like a charm