serialport/node-serialport #1818: Not Installing on Electron & Ubuntu

dan-sam, 2019-03-15:
### Summary of Problem

(Please answer all 3)

1) I have an Up Board: [https://up-board.org/up/specifications/](url)

2) I have installed Ubuntu 18.04 on it according to their documentation:
[https://wiki.up-community.org/Ubuntu](url) (scroll down to Ubuntu 18.04 installation and configuration)

4) Please take note that for this Up Board I had to install Ubuntu kernel 4.15.0 for UP from PPA on Ubuntu 18.04. ==> It's all on the same page (wiki link above)
This may be an important detail for this issue.

5) I installed Electronjs and my application is running just fine. So far so good. I then have: 

- installed electron-rebuild

- added it to my json file ==> “install” : “electron-rebuild",

- I run npm install ==> everything good so far

Btw: node version: v9.11.11

6) I installed serialport ==> npm i --save serialport

7) I run again ==> npm install ==> ERROR (Log shown below)

I managed to do this process 100 times on Ubuntu and even on this Board. But when I successfully did it on this board I installed Ubuntu without their Kernel changing instructions as described here ==> [https://wiki.up-community.org/Ubuntu](url) (same link as above)

So my only guess right now is that something is missing here because of these specific changes. But unfortunately I don't really get what it is. Can anyone help please?

========================THE ERROR MSG============================

`An unhandled error occurred inside electron-rebuild
make: Entering directory '/home/dan/Desktop/samapp/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
In file included from ../../../nan/nan_converters.h:67:0,
                 from ../../../nan/nan.h:222,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
../../../nan/nan_converters_43_inl.h: In static member function ‘static Nan::imp::ValueFactoryBase<bool>::return_t Nan::imp::ToFactory<bool>::convert(v8::Local<v8::Value>)’:
../../../nan/nan_converters_43_inl.h:60:46: error: no matching function for call to ‘v8::Value::BooleanValue(v8::Isolate*&)’
   return Just<bool>(val->BooleanValue(isolate));                               \
                                              ^
In file included from /home/dan/.electron-gyp/4.0.7/include/node/node.h:67:0,
                 from ../../../nan/nan.h:53,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/dan/.electron-gyp/4.0.7/include/node/v8.h:2580:37: note: candidate: v8::Maybe<bool> v8::Value::BooleanValue(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT Maybe<bool> BooleanValue(Local<Context> context) const;
                                     ^~~~~~~~~~~~
/home/dan/.electron-gyp/4.0.7/include/node/v8.h:2580:37: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘v8::Local<v8::Context>’
In file included from /home/dan/.electron-gyp/4.0.7/include/node/v8.h:26:0,
                 from /home/dan/.electron-gyp/4.0.7/include/node/node.h:67,
                 from ../../../nan/nan.h:53,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/dan/.electron-gyp/4.0.7/include/node/v8.h:2588:47: note: candidate: bool v8::Value::BooleanValue() const
   V8_DEPRECATE_SOON("Use maybe version", bool BooleanValue() const);
                                               ^
/home/dan/.electron-gyp/4.0.7/include/node/v8config.h:346:48: note: in definition of macro ‘V8_DEPRECATE_SOON’
 #define V8_DEPRECATE_SOON(message, declarator) declarator
                                                ^~~~~~~~~~
/home/dan/.electron-gyp/4.0.7/include/node/v8.h:2588:47: note:   candidate expects 0 arguments, 1 provided
   V8_DEPRECATE_SOON("Use maybe version", bool BooleanValue() const);
                                               ^
/home/dan/.electron-gyp/4.0.7/include/node/v8config.h:346:48: note: in definition of macro ‘V8_DEPRECATE_SOON’
 #define V8_DEPRECATE_SOON(message, declarator) declarator
                                                ^~~~~~~~~~
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:41:75: warning: ‘v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)’ is deprecated: Use Isolate version [-Wdeprecated-declarations]
[... 93 lines cut ...]
gyp ERR! not ok 

Failed with exit code: 1
    at SafeSubscriber._error (/home/dan/Desktop/samapp/node_modules/spawn-rx/lib/src/index.js:267:84)
    at SafeSubscriber.__tryOrUnsub (/home/dan/Desktop/samapp/node_modules/rxjs/internal/Subscriber.js:207:16)
    at SafeSubscriber.error (/home/dan/Desktop/samapp/node_modules/rxjs/internal/Subscriber.js:158:26)
    at Subscriber._error (/home/dan/Desktop/samapp/node_modules/rxjs/internal/Subscriber.js:92:26)
    at Subscriber.error (/home/dan/Desktop/samapp/node_modules/rxjs/internal/Subscriber.js:72:18)
    at MapSubscriber.Subscriber._error (/home/dan/Desktop/samapp/node_modules/rxjs/internal/Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (/home/dan/Desktop/samapp/node_modules/rxjs/internal/Subscriber.js:72:18)
    at SafeSubscriber._next (/home/dan/Desktop/samapp/node_modules/spawn-rx/lib/src/index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (/home/dan/Desktop/samapp/node_modules/rxjs/internal/Subscriber.js:207:16)
    at SafeSubscriber.next (/home/dan/Desktop/samapp/node_modules/rxjs/internal/Subscriber.js:145:22)
npm ERR! code ELIFECYCLE
npm ERR! errno 255
npm ERR! electron-quick-start@1.0.0 install: `electron-rebuild`
npm ERR! Exit status 255
npm ERR! 
npm ERR! Failed at the electron-quick-start@1.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/dan/.npm/_logs/2019-03-14T22_44_50_163Z-debug.log

`

dan-sam, 2019-03-18:
I switched to Node Version v10.13.0 and this solved the problem. 
It seems that v11.11.0 probably triggers the problem.

==> Is this issue known?

If someone else runs into this issue and you are just starting off: use `nvm` to install node version v10.13.0.

dsteinman, 2019-04-10:
I ran into this problem also.  I think Serialport doesn't play well with Electron 4.x because there's no "binding.gyp" in the root of the project, which (I think) Electron requires to rebuild the library for it's version of Node.

**gyp: binding.gyp not found (cwd: /Users/dstein/electron-app/node_modules/serialport) while trying to load binding.gyp**

```
.gyp info it worked if it ends with ok
 gyp info using node-gyp@3.6.2
 gyp info using node@11.4.0 | darwin | x64
 gyp info spawn /usr/local/bin/python2
 gyp info spawn args [ '/Users/dstein/.config/yarn/global/node_modules/node-gyp/gyp/gyp_main.py',
 gyp info spawn args   'binding.gyp',
 gyp info spawn args   '-f',
 gyp info spawn args   'make',
 gyp info spawn args   '-I',
 gyp info spawn args   '/Users/dstein/electron-app/node_modules/serialport/build/config.gypi',
 gyp info spawn args   '-I',
 gyp info spawn args   '/Users/dstein/.config/yarn/global/node_modules/node-gyp/addon.gypi',
 gyp info spawn args   '-I',
 gyp info spawn args   '/Users/dstein/.electron-gyp/.node-gyp/4.1.4/include/node/common.gypi',
 gyp info spawn args   '-Dlibrary=shared_library',
 gyp info spawn args   '-Dvisibility=default',
 gyp info spawn args   '-Dnode_root_dir=/Users/dstein/.electron-gyp/.node-gyp/4.1.4',
 gyp info spawn args   '-Dnode_gyp_dir=/Users/dstein/.config/yarn/global/node_modules/node-gyp',
 gyp info spawn args   '-Dnode_lib_file=/Users/dstein/.electron-gyp/.node-gyp/4.1.4/<(target_arch)/node.lib',
 gyp info spawn args   '-Dmodule_root_dir=/Users/dstein/electron-app/node_modules/serialport',
 gyp info spawn args   '-Dnode_engine=v8',
 gyp info spawn args   '--depth=.',
 gyp info spawn args   '--no-parallel',
 gyp info spawn args   '--generator-output',
 gyp info spawn args   'build',
 gyp info spawn args   '-Goutput_dir=.' ]
 gyp: binding.gyp not found (cwd: /Users/dstein/electron-app/node_modules/serialport) while trying to load binding.gyp
 gyp ERR! configure error 
 gyp ERR! stack Error: `gyp` failed with exit code: 1
 gyp ERR! stack     at ChildProcess.onCpExit (/Users/dstein/.config/yarn/global/node_modules/node-gyp/lib/configure.js:336:16)
 gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
 gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:254:12)
 gyp ERR! System Darwin 18.2.0
 gyp ERR! command "/usr/local/bin/node" "/usr/local/bin/node-gyp" "rebuild" "--target=4.1.4" "--arch=x64" "--dist-url=https://atom.io/download/electron"
 gyp ERR! cwd /Users/dstein/electron-app/node_modules/serialport
 gyp ERR! node -v v11.4.0
 gyp ERR! node-gyp -v v3.6.2
 gyp ERR! not ok
```

reconbot, 2019-04-10:
We do have a bindings.gyp
---
Francis Gulotta
wizard@roborooter.com

dsteinman, 2019-04-10:
I see them here in the repo https://github.com/node-serialport/node-serialport/tree/master/packages/bindings

But when I do `npm install serialport` the file isn't there, so Electron cannot rebuilt the module:

```
drwxr-xr-x     8 dstein  staff    256 Apr 10 10:25 .
drwxr-xr-x  1096 dstein  staff  35072 Apr 10 10:25 ..
-rw-r--r--     1 dstein  staff  35669 Oct 26  1985 CHANGELOG.md
-rw-r--r--     1 dstein  staff   1105 Oct 26  1985 LICENSE
-rw-r--r--     1 dstein  staff    181 Oct 26  1985 README.md
drwxr-xr-x     4 dstein  staff    128 Apr 10 10:25 lib
-rw-r--r--     1 dstein  staff   2435 Apr 10 10:25 package.json
-rw-r--r--     1 dstein  staff    230 Oct 26  1985 test.js
```

I guess maybe the contents of that bindings directory would have to be added to the NPM module.

reconbot, 2019-04-10:
It’s a different package

dsteinman, 2019-04-10:
Oh, perhaps maybe this may be part of the problem.  I'm not sure what the exact solution is, but I suspect that because the bindings.gyp is not in the main "serialport" npm module, that this may be the reason for this error?

**gyp: binding.gyp not found (cwd: /Users/dstein/electron-app/node_modules/serialport) while trying to load binding.gyp**

Maybe that bindings package would need to be combined for this to work.

closed 2019-05-20 by reconbot