serialport/node-serialport #1819: Exporting as a module help

callen5914, 2019-03-16:
### Summary of Support Request

_Much of this request for help stems from my lack of understanding of the docs and JS_

- [x] Establish a SerialPort connection
- [x] Can Read Data coming from SerialPort
- [ ] Once the port is established, export or use `port` in other JS files to issue commands to an Arduino Compatible device

### Existing Code

**_File that would establish the port connection_**

```js
const SerialPort = require('serialport')
const Readline = require('@serialport/parser-readline')

// App level settings
let appPortSetting = document.getElementById('appPort')
// let appPortSetting = toString(document.getElementById('appPort'))
let appBaudrateSetting = document.getElementById('appBaudrate')
const appConnectBtn = document.getElementById('appConnectBtn')

// Is SerialPort connected
let isSerialConnected = true

// Array of ports (May get rid of this later)
let portList = []

let port

// Get all serial ports (May no longer need this)
function getPorts() {
    SerialPort.list((err, ports) => {
        if (err) { console.log('No ports to list') }

        ports.forEach((port) => {
            portList.push(port.comName)
        })
    })
    return portList
}

// Establish port connection
appConnectBtn.addEventListener('click', () => {
    port = new SerialPort(appPortSetting.innerText, { baudRate: Number(appBaudrateSetting.innerText) }, (err) => {
        if (err) {
            console.log(err.message)
            appConnectBtn.innerHTML = 'Click to retry'
            isSerialConnected = false
        } else {
            appConnectBtn.innerHTML = 'Connected to Port'
            isSerialConnected = true
            listen()
            return port
        }
    })

})

// Listen to port once connected
listen = function () {
    const parser = port.pipe(new Readline({ delimiter: '\r\n' }))
    parser.on('data', (data) => { console.log('Message: ', data) })
    port.on('close', (err) => {
        if (err.disconnected === true) { 
            console.log('port disconnected')
            appConnectBtn.innerHTML = 'Click to reconnect'
            isSerialConnected = false
        }
    })
}

// Instantiate
getPorts()

module.exports = { getPorts, port }
// Code
```

The function to `get_ports` helps me to generate options for a select box in the dom. I do have a hunch that way I am creating the `port` is not correct here, because I have to use the `listen` function to listen and parse responses and such. I feel like this is hurting my code's modularity.

One thing to note is that a portion of the DOM changes depending on where the user is in the application. So the JS below is called into play in a different portion of the app. I would like to make it that when a button is clicked, a `port.write()` could happen.

_**File that I would like to send a command to an Arduino Compatible**_

```js
const SerialPort = require('serialport')
const ports = require('./kuristo-ports')

// Getting the elements from the DOM needed to write to serial port
const kuristoWriteBtn = document.getElementById('kuristo-write-btn')
const kuristoNewTagLabel = document.getElementById('kuristo-new-tag-label')

kuristoWriteBtn.addEventListener('click', () => {
    console.log('Would write to tag')
    ports = ports.port
    console.log(ports)
})
```

### Versions, Operating System and Hardware

```js
"dependencies": {
    "electron": "^4.0.7",
    "electron-rebuild": "^1.8.4",
    "electron-settings": "^3.2.0",
    "electron-shortcut-normalizer": "^1.0.0",
    "glob": "^7.1.3",
    "highlight.js": "^9.15.6",
    "serialport": "^7.1.4"
  }
```

callen5914, 2019-04-06:
Any help?

reconbot, 2019-04-08:
What are you asking? What doesn't work?

callen5914, 2019-04-09:
So if I have a file, let's call it main.js that establishes a connection using SerialPort. If we called it in the conventional way:

`port = new SerialPort`

How I go about calling `port` in a separate js file, for example:

`const port = require('main.js/port')`

I tried using module exports as listed above. But it never seems to work. I was hoping to write my file structure with a separation of concerns. Meaning instead of having one big long file, I could have a file that establish the connection and that connections could be called upon for my various serial port actions.

In my particular use case, I am issuing various commands to an Arduino with an RFID shield.

Cheers,

JohnzBallad, 2019-04-16:
I'm having the same issue. Literally the same. I want to use the SerialPort in another JS file, same instance, same connection. Arduino environment too.

reconbot, 2019-04-23:
That's not how modules work in nodejs. You can try.
```
const port = require('./main').port
```

closed 2019-06-29 by stale[bot]