serialport/node-serialport #1822: "Cannot lock port" if it was previously opened before NWJS page changed

NorbarTDB, 2019-03-19:
### Summary of Problem

Hi there! I'm using Node Serialport with NWJS, and I've noticed that if I open a serial port and navigate to a new page in my UI, any subsequent attempts to use that serial port result in `Resource temporarily unavailable Cannot lock port` errors. This makes sense, since I left it open before.

If I close my application and re-open it, then I can use the serial port again. Or if I close the port prior to navigating, same thing when I load the next page. That makes sense too. So I simply try to close it beforehand such that I can re-open it on the new page.

However I'd like a way to leave a port open between pages and re-acquire it on the new page? I don't want another application calling dibs on the released port while I'm changing pages. I've not found a means to do this in the SerialPort documentation yet, so any suggestions would be much appreciated! 

Since my application is the thing that opened the port in the first place, is there a way for me to re-acquire it once I've changed pages?

### Versions, Operating System and Hardware

- SerialPort@? 2.0.6
- Node.js v? 8.10
- NWJS? 0.35.4
- Ubuntu
- Any USB serial device

NorbarTDB, 2019-04-04:
Having navigated away from a page using SerialPort (without closing the port, as described above), I can see the application still has a hold of the port somehow:

lsof | grep tty shows the processes from the application as having the port open:

```
exe       16370                  user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
TaskSched 16370  2197            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
TaskSched 16370  2198            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
exe       16370  9068            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
exe       16370  9069            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
exe       16370  9070            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
exe       16370  9071            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
ScriptStr 16370 12948            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
TaskSched 16370 16372            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
TaskSched 16370 16374            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
Chrome_Ch 16370 16375            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
GpuMemory 16370 16376            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
Font_Prox 16370 16377            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
Composito 16370 16378            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
Composito 16370 16379            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
Composito 16370 16380            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
Composito 16370 16381            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
TaskSched 16370 16893            user  149uW     CHR              188,0          0t0        550 /dev/ttyUSB0
```

How do I get this to be released? Does SerialPort expose a way to release ports it technically already has a grip on?

NorbarTDB, 2019-04-29:
Still haven't found a way around this! Has anyone got any suggestions for releasing locked ports when using Node Serialport via NWJS?

reconbot, 2019-05-16:
I know we work work alright with electron, I can't give good advise for NWJS. In electron you'd use the main process to open the port not the render process.

closed 2019-07-22 by stale[bot]