serialport/node-serialport #1826: SerialPort write not working

allahyar, 2019-03-29:
Hi, I'm tray to send string data from pc to arduino, Opened port and write message but it does not work  :(

```js
const SerialPort = require('serialport');
const Readline = require('@serialport/parser-readline');
const port = new SerialPort('COM7', {baudRate: 9600});

const parser = new Readline({delimiter: '\n'});
port.pipe(parser);

SerialPort.list((error, ports) => {
    if (ports) console.log(`connected to ${ports[1].manufacturer}`);
});

port.on('open', () => console.log(`port open. Data rate: ${port.baudRate}`));
port.write('0\n', (err) => {
    if (err) return console.log('Error on write: ', err.message);
    console.log('message written')
});```

// Code
```void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(9600); 
}

void loop() {
   if(Serial.available()){
    switch(Serial.read()){
      case '0': digitalWrite(LED_BUILTIN, LOW); break;
      case '1': digitalWrite(LED_BUILTIN, HIGH); break;
      default: break;
    }
  }
}```

- SerialPort@ 7.1.4
- Node.js 10.4.2
- Windows
- Hardware and chipset Arduino Mega 2560

jonathangaldino, 2019-04-25:
I'm having the same problem.

It seems that when I use the #write function, it is not always that it writes the information on the serial port and, in case of not writing, the Error is not released.

I don't know why tho. But your scenario looks just like mine.

reconbot, 2019-04-25:
It's usually wrong baudrates if you don't get any errors.

jonathangaldino, 2019-04-26:
I did what you asked. I debugged and I couldn't find the reason why it is not writing. I'll leave my code along with a short output from the debug, below. 

I noticed something: somethimes it does write. And I can see the debug using bindings/unixWrite, but it's rare. It seems like something related to the time that the js script is writing. I don't know.

My debug:

>❯ DEBUG=serialport/binding* yarn start
yarn run v1.13.0
$ node src/index.js
  serialport/bindings loading DarwinBinding +0ms
  serialport/binding-abstract open +0ms
  serialport/binding-abstract open +3ms
Error:  Error Resource temporarily unavailable Cannot lock port
  serialport/bindings/poller Creating poller +0ms
Connection with undefined opened.
  serialport/binding-abstract read +4ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +1ms
  serialport/bindings/poller received "readable" +7s
  serialport/binding-abstract read +7s
  serialport/bindings/unixRead Starting read +7s
  serialport/bindings/unixRead Finished read 2 bytes +0ms
  serialport/binding-abstract read +2ms
  serialport/bindings/unixRead Starting read +2ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +4ms
  serialport/bindings/poller received "readable" +0ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixRead Starting read +1ms
  serialport/bindings/unixRead Finished read 4 bytes +0ms
  serialport/binding-abstract read +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +0ms
  serialport/bindings/poller received "readable" +4ms
  serialport/binding-abstract read +4ms
  serialport/bindings/unixRead Starting read +4ms
  serialport/bindings/unixRead Finished read 4 bytes +0ms
  serialport/binding-abstract read +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +2ms
  serialport/bindings/poller received "readable" +2ms
  serialport/binding-abstract read +4ms
  serialport/bindings/unixRead Starting read +3ms
  serialport/bindings/unixRead Finished read 4 bytes +0ms
  serialport/binding-abstract read +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +0ms
  serialport/bindings/poller received "readable" +4ms
  serialport/binding-abstract read +4ms
  serialport/bindings/unixRead Starting read +4ms
  serialport/bindings/unixRead Finished read 4 bytes +0ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixRead Starting read +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +1ms
  serialport/bindings/poller received "readable" +3ms
  serialport/binding-abstract read +3ms
  serialport/bindings/unixRead Starting read +3ms
  serialport/bindings/unixRead Finished read 3 bytes +1ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +1ms
  serialport/bindings/poller received "readable" +3ms
  serialport/binding-abstract read +3ms
  serialport/bindings/unixRead Starting read +4ms
[... 303 lines cut ...]
      lastSyncMillis = currentMillis;
      break;
    }
  }

  const size_t bufferSize = JSON_OBJECT_SIZE(3) + 120;
  StaticJsonDocument<bufferSize> doc;

  DeserializationError err = deserializeJson(doc, Serial);

  if (err)
  {
    digitalWrite(LED_BUILTIN, HIGH);
    Serial.flush();
    return;
  }

  updateCycle = doc["updateCycle"];
  minSetpoint = doc["min"];
  maxSetpoint = doc["max"];

  digitalWrite(LED_BUILTIN, LOW);
}

```

reconbot, 2019-04-26:
Try passing lock: false

jonathangaldino, 2019-04-26:
> Try passing lock: false
> -- --- Francis Gulotta wizard@roborooter.com

Thank you for the replies. I think it solved my problem.

Can you explain how this lock system works and why it wasn't letting the write function work properly?

reconbot, 2019-04-26:
The log was was saying something about the lock failing which should cause
an error. I'm on mobile so I haven't dug in.

Your drivers, or other devices using the port may prevent the locking
behavior, but it shouldn't stop the port from opening without an error.

jonathangaldino, 2019-04-26:
> The log was was saying something about the lock failing which should cause an error. I'm on mobile so I haven't dug in. Your drivers, or other devices using the port may prevent the locking behavior, but it shouldn't stop the port from opening without an error.
> […](#)

Understood. Thanks, tho. Means a lot.

vlondonoo1, 2020-07-24:
@reconbot  I'm not sure  if  I'm having a similar problem,  it's because I'm getting this message

```
 serialport/stream flush attempted, but port is not open +8ms
  serialport/stream _read queueing _read for after open +3ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/ttyUSB0 +3ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +1ms
  serialport/binding-abstract read +15ms
  serialport/stream _write 15 bytes of data +1ms
  serialport/binding-abstract write 15 bytes +1ms
  serialport/bindings/unixWrite Starting write 15 bytes offset 0 bytesToWrite 15 +0ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixWrite write returned: wrote 15 bytes +1ms
  serialport/bindings/unixWrite Finished writing 15 bytes +0ms
  serialport/stream binding.write write finished +2ms
  serialport/stream drain +0ms
  serialport/binding-abstract drain +2ms
message written
  serialport/bindings/unixRead read error { [Error: EAGAIN: resource temporarily unavailable, read] errno: -11, code: 'EAGAIN', syscall: 'read' } +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +5ms
  serialport/stream binding.drain finished +1ms
```

I'm just writing a message, and the first time that it writes after opening the port It's getting a response, I mean reading the message, then It writes it again and it never gets another response from the device, I'm not sure if it's about reading, writing or related to the port(open/close)

rpd-512, 2021-01-04:
> It's usually wrong baudrates if you don't get any errors. You can try debug mode if you want to see what's going on under the hood. https://serialport.io/docs/en/guide-debugging

The link isn't working :(

reconbot, 2021-01-04:
Sorry I broke the links with the last site upgrade. I dropped the language support because it was mostly untranslated.

https://serialport.io/docs/guide-debugging

PrathameshTM, 2021-02-02:
I had this problem where it was not writing to the port.
I introduced a delay of 4 seconds before writing and it started working.

I tried writing `port.on('open')` event but it didn't work.

josegpulido, 2021-03-09:
I've an Arduino sketch that prints a counter number which increments on every loop, working properly on Arduino IDE's Serial Monitor until infinity, but stopping at number 967 using the serialport library. Why this is happening? I'm using the Ready and Readline (to user 'data' event transformed by the 'delimiter') parsers properly, following the docs examples. It freezes even just using the on 'data' directly from the port.

benswinburne, 2021-04-27:
> I had this problem where it was not writing to the port.
> I introduced a delay of 4 seconds before writing and it started working.
> 
> I tried writing `port.on('open')` event but it didn't work.
> 
> Is there any other elegant way of doing this?

This is the only way I can get my application to send successfully too, a somewhat arbitrary delay of 2000ms. Much lower and it stops working.

@reconbot presumably this is not the intended behaviour and `.on('open')` should be sufficient to have a usable port?

benswinburne, 2021-04-27:
Just to expand on the on the above

Simple Node App

```
    const SerialPort = require('serialport');
    const Readline = require('@serialport/parser-readline');

	const writeDelay = 2000; // reducing this value stops it working
	const path = '/dev/tty.usbmodem113401';
	const port = new SerialPort(path, {
		lock: false,
		baudRate: 9600,
	});

	port.on('open', () => {
		setTimeout(function () {
			console.log('sending: ping');
			port.write('ping');
		}, writeDelay);
	});

	const parser = new Readline({ delimiter: '\r\n' });

	port.pipe(parser);

	parser.on('data', function (data) {
		console.log('from arduino:', data);
	});
```

Arduino Code

```
void setup() {
  Serial.begin(9600);
}

void loop() {
  while (Serial.available() > 0) {
    String str = Serial.readString();

    if (str == "ping") {
      Serial.println("pong");
      Serial.flush();
    }
  }
}
```

Debug with 2000ms delay (Working)

```
  serialport/bindings loading DarwinBinding +0ms
started
  serialport/stream opening path: /dev/tty.usbmodem113401 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +2ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/tty.usbmodem113401 +5ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +1ms
  serialport/binding-abstract read +9ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -35,
  code: 'EAGAIN',
  syscall: 'read'
} +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +3ms
sending: ping
  serialport/stream _write 4 bytes of data +2s
  serialport/binding-abstract write 4 bytes +2s
  serialport/bindings/unixWrite Starting write 4 bytes offset 0 bytesToWrite 4 +0ms
  serialport/bindings/unixWrite write returned: wrote 4 bytes +2ms
  serialport/bindings/unixWrite Finished writing 4 bytes +0ms
  serialport/stream binding.write write finished +4ms
  serialport/bindings/poller received "readable" +3s
  serialport/bindings/unixRead Starting read +3s
  serialport/bindings/unixRead Finished read 6 bytes +1ms
  serialport/stream binding.read finished { bytesRead: 6 } +1s
from arduino: pong
  serialport/stream _read reading { start: 6, toRead: 65530 } +3ms
  serialport/binding-abstract read +1s
  serialport/bindings/unixRead Starting read +3ms
  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -35,
  code: 'EAGAIN',
  syscall: 'read'
} +1ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +5ms
```

Debug with 1000ms delay (Not Working)

```
  serialport/bindings loading DarwinBinding +0ms
started
  serialport/stream opening path: /dev/tty.usbmodem113401 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +2ms
  serialport/bindings/poller Creating poller +0ms
  serialport/stream opened path: /dev/tty.usbmodem113401 +42ms
  serialport/stream _read reading { start: 0, toRead: 65536 } +1ms
  serialport/binding-abstract read +47ms
  serialport/bindings/unixRead Starting read +0ms
  serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -35,
  code: 'EAGAIN',
  syscall: 'read'
} +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +6ms
sending: ping
  serialport/stream _write 4 bytes of data +1s
  serialport/binding-abstract write 4 bytes +999ms
  serialport/bindings/unixWrite Starting write 4 bytes offset 0 bytesToWrite 4 +0ms
  serialport/bindings/unixWrite write returned: wrote 4 bytes +1ms
  serialport/bindings/unixWrite Finished writing 4 bytes +0ms
  serialport/stream binding.write write finished +2ms
```

toncho11, 2023-09-15:
> I had this problem where it was not writing to the port. I introduced a delay of 4 seconds before writing and it started working.
> 
> I tried writing `port.on('open')` event but it didn't work.
> 
> Is there any other elegant way of doing this?

I had to add at least 1200 ms of delay before sending a character to make it work. I am using a 10 years old Arduino UNO. The 400 ms in the docs are actually not a good estimation. As we can see it can go to 1200 ms or 4000 ms.