serialport/node-serialport #1828: Is there support for RS-422 on linux?

jrj2211, 2019-04-02:
### Summary of Problem

I am trying to connect to a 422 serial port and am curious if this library has support for it. We recently switched from 232 to 422, and we could successfully connect to our device before with 232. When we now try to connect, the .open(function(err)) callback just tells that its opening the port but nothing else ever happens.

I did notice 2 years ago there was a feature request to add a wrapper for 485 (issue #903). Not sure if that went anywhere and if there isnt 485, I'm not sure if that also means there isnt 422.

### Code to Reproduce the Issue

```js
const SerialPort= require('./classes/RiverInterface.js');
riverInterface = new SerialPort("/dev/ttts2", {baudRate: 115200});
```

### Versions, Operating System and Hardware
OS: Linux - Debian 9
System: Technologic Systems 7800

reconbot, 2019-04-03:
We don't support it in particular, I haven't done any research into what would be needed either.

jrj2211, 2019-04-03:
Okay thank you for the information.

closed 2019-04-03 by jrj2211

reconbot, 2019-04-04:
I'm ok with supporting it but would need someone to do the research on how to support it cross platform