serialport/node-serialport #1829: Docker: serialport invalid ELF header

lowgos, 2019-04-03:
### Summary of Problem

- What are you trying to do?
      → Dockerize my nodejs API containing colinbdclark/osc.js
- What happens?
      → serialport can't install succesfully making my nodejs app crash at launch
- What should have happened?
      → serialport which is a dependency to colinbdclark/osc.js should install without errors

### Code to Reproduce the Issue

```Docker
FROM node:stretch

# Create app directory
WORKDIR /usr/src/app

# Install app dependencies
COPY package.json ./
COPY yarn.lock ./

RUN yarn install

# Bundle app source
COPY . .

# Default port in the docker file
EXPOSE 8080

## Launch the wait tool and then your application
CMD yarn run start
```

### Error

```
Error: /usr/src/app/node_modules/serialport/build/Release/serialport.node: invalid ELF header
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:846:18)
    at Module.load (internal/modules/cjs/loader.js:672:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:612:12)
    at Function.Module._load (internal/modules/cjs/loader.js:604:3)
    at Module.require (internal/modules/cjs/loader.js:711:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at bindings (/usr/src/app/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/usr/src/app/node_modules/serialport/lib/bindings/linux.js:2:36)
    at Module._compile (internal/modules/cjs/loader.js:805:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:816:10)
    at Module.load (internal/modules/cjs/loader.js:672:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:612:12)
    at Function.Module._load (internal/modules/cjs/loader.js:604:3)
    at Module.require (internal/modules/cjs/loader.js:711:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at Object.<anonymous> (/usr/src/app/node_modules/serialport/lib/bindings/auto-detect.js:16:22)
```

### Versions, Operating System and Hardware

- serialport@6.2.2
- Node.js v10.15.1
- Developping on macOS (fine) and Docker on Debian using the node:stretch image

### Bonus

I can't seem to find how to fix the issue.
I began Dockerizing with node:alpine but this image is too bare for serialport it seems. I had a lot of building issue and I would like to at least test with node:stretch.
I think I'm not mistaken by posting this here instead of colinbdclark/osc.js because it seems it's serialport that is crashing.

I hope the absence of code isn't forbidden. I don't use serialport but osc and it crashes before any kind of code seems to be executed.

I hope you can help me pinpoint exactly what I do that I shouldn't or how to make serialport work seamlessly. I think making serial ports working inside a Docker isn't a good idea but osc is really fine for my app 😅

Thank you for your time, Gabriel

lowgos, 2019-04-05:
Hey, I got some news, it works perfectly on Debian. Docker for Mac is some serious BS. I won't search for an answer anymore but I don't close the issue myself maybe someone is gonna put some news on how to actually solve this.
Hope it can help somebody ;)

reconbot, 2019-04-28:
That's great to hear! I haven't had issues when using npm and docker for mac however. I don't know what yarn keeps in it's lock files with regards to platform. Check your logs to see if it's downloading for the wrong platform?

reconbot, 2019-05-02:
I'm going to guess

```
# Bundle app source
COPY . .
```
overwrites your docker installed `node_modules` with the OSX version which would have a osx binary and not a linux one.

closed 2019-05-02 by reconbot