serialport/node-serialport #1830: SerialPort 7.1.4 crashing on Win64 when calling list()

caffeinatedbits, 2019-04-04:
### Summary of Problem
My application gets a list of serial ports, and then makes a connection to a device to transfer data. At the point in which serialport .list() is called, the application is crashing with Crash reason:  EXCEPTION_ACCESS_VIOLATION_READ in bindings.node (dumps attached). I've got this app installed on ~500 end user systems, but the error is only occurring on one, a laptop with Win10 Pro x64.

(Please answer all 3)
[dump1.txt](https://github.com/node-serialport/node-serialport/files/3045496/dump1.txt)
[dump2.txt](https://github.com/node-serialport/node-serialport/files/3045497/dump2.txt)
[dump3.txt](https://github.com/node-serialport/node-serialport/files/3045498/dump3.txt)

- What are you trying to do?
List serial ports

- What happens?
Application crashes (dump generated)

- What should have happened?
Serial ports should be listed

### Code to Reproduce the Issue

```js
const SerialPort = new.require('serialport');
SerialPort.list().then( (ports) => { ...code... }, (error) => { ...code... });

// Code
```

### Versions, Operating System and Hardware

- SerialPort@7.14
- Node.js v? - 11.10.0 / NWJS 0.36.3
- Windows? Linux? Mac? - Windows 10 x64
- Hardware and chipset? (Prolific/FTDI/Other) - Unknown USB-to-RS232 Serial Adapter

caffeinatedbits, 2019-04-15:
Update: I've now discovered a total of 3 systems with this same issue, and they all have a Brother multifunction printer attached by USB. When the printer cable is disconnected, my app runs flawlessly. Why would a Brother printer connected via USB cause issues with serialport?

reconbot, 2019-04-15:
It would be the brother print drivers that don't like to be querried for device info. We use USB information to get more info on each serialport.

reconbot, 2019-05-02:
You can try the [`usb`](https://www.npmjs.com/package/usb) npm package to see if you can find your device in a way that wont mess up the printers. I'd also file a bug with Brother.

closed 2019-05-02 by reconbot