serialport/node-serialport #1831: Allow setting no baudRate at all as some devices don't allow setting one

Thomas12, 2019-04-06:
Hi,

I have a device (USB device that behaves like a serial port) that results in ioctl error when setting a baud rate.

Error: Inappropriate ioctl for device setting custom baud rate of 9600.
(Happens with any baudrate).

If I disable the setBaudRate-fcuntion in the .c source, then the device works.
Could you please add a functionality that e.g. setting baudRate to "0", then
the baudRate is never set?

Thank you!

Thomas

reconbot, 2019-04-27:
Can you tell me more about these devices? Can I buy one?

matheuspl, 2019-11-26:
I have this issue too. I'm using a usb matricial printer Epson FX-890, the custom path for usb printers is /dev/usb/lpx, where x is a number. After searching a lot, found some informations:

/dev/usb/lpx - printing to USB printer, where x is the printer number. /dev/usb/lp0 usually
/dev/ttyUSBx - printing to printer connected using USB2Serial adapter
/dev/ttySx - printing to direct RS232 attached printer

USB printers protocol is diferent from the others and does not accept baud rate configurations like the others. If you try to change the baud rate directly with this command will get the same error:

stty -F /dev/usb/lp0 19200

afzaal-ameer, 2021-04-17:
Any workaround for this ?

skylarmt, 2025-02-14:
I'm also running into this with a Pi Pico. It shows up as `/dev/ttyACMx` but I'm using udev rules to add a symlink that identifies it by serial number. The `/dev/ttyACM0` path works, but the symlink path does not, giving this same ioctl error.

This same software project has also run into the printer issue @matheuspl had. I ended up solving it by opening the printer paths as regular files and saving the print data to them. However, this means I can't easily read any responses or error codes (like "paper out") from the printer.