serialport/node-serialport #1832: Can not use cctalk parser

scorpioRED, 2019-04-10:
### Summary of Problem
Hi! I try to use cctalk parser, but unsuccessful.
(Please answer all 3)

- What are you trying to do?
Hi! I try to use cctalk parser, but unsuccessful.

- What happens?
 I got an error

```
const parser = port.pipe(new CCtalk())
                             ^

ReferenceError: CCtalk is not defined
    at Object.<anonymous> (/home/user/tst/custom/index.js:11:30)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:188:16)
    at bootstrap_node.js:609:3
```

- What should have happened?

### Code to Reproduce the Issue

```js
const SerialPort = require("serialport")
const CCTalk = require('@serialport/parser-cctalk')

const serialport = new SerialPort('/dev/ttyUSB0')

const parser = port.pipe(new CCtalk())

parser.on('data', console.log)

port.open((err) => {
    if (err) {
        return console.log("Error opening port: ", err.message);
    } else {
        console.log('port has ben opened')
        port.write(Buffer.from("2012458", "hex"), function(err){
                console.log('writing error => ', err)
        })
    }

});

port.on("data", (data) => {
        console.log('data => ', data)
})

port.on("error", function(err) {
  console.log('Error: ', err.message)
})

```

### Versions, Operating System and Hardware

- SerialPort@?
  "serialport": "^7.1.4"

- Node.js v?
   v8.10.0

- Windows? Linux? Mac?
   Linux Lubuntu
- Hardware and chipset? (Prolific/FTDI/Other)

reconbot, 2019-04-10:
This looks like a nodejs issue, what happens if you try the require in the
repl?

scorpioRED, 2019-04-10:
> This looks like a nodejs issue, what happens if you try the require in the repl?
> -- --- Francis Gulotta wizard@roborooter.com

`const repl = require('repl');` works fine

reconbot, 2019-04-10:
I mean can you use the node repl to require the package and see what you
get. It should just be the class.
---
Francis Gulotta
wizard@roborooter.com

krutkay, 2019-04-11:
@scorpioRED I think it may just be a capitalisation issue... try changing to `new CCTalk()` instead of `new CCtalk()`

reconbot, 2019-04-27:
That looks right, I also recommend using https://eslint.org/ which helps automatically highlight these things.

closed 2019-04-27 by reconbot