serialport/node-serialport #1835: Buffer overrun(?) when using ByteParser with long transfers and busy main loop

danni, 2019-04-15:
I can reliably reproduce a strange problem during data transfer when the mainloop is quite busy (including when the devtools are open). Making the mainloop busier makes it occur sooner in the transfer.

```
const port = new SerialPort( ... )
const parser = port.pipe(new ByteLength({ length: 4096 });

parser.on('data', frame => parse(frame))
// or
for await (const frame of parser)
```

If the app is sufficiently idle, this will complete. If things are getting busy (e.g. simultaneous transfers, UI updates, encoding the output), then I will eventually slip off the framing and end up 10-40 bytes offset in my next callback. It feels like I'm blowing a buffer somewhere in serialport, but I can't find it.

reconbot, 2019-05-02:
what's parse? Generally mixing the data event (flowing mode) with async iterators (pull mode) isn't a good idea.

danni, 2019-05-02:
`parse` is parsing the binary data. I could rewrite this as a `Transform` stream, but the very next step requires splitting into multiple streams to demultiplex the binary. Waiting and pushing is clearer.

I worked around the problem by decreasing my transfer window size and it seems to be working well. I also rewrote it to remove async iterators, because it was hard to manage timeouts. A warning about blowing your buffer would be valuable though.

reconbot, 2019-05-02:
We don't keep a buffer that can get blown, I think you're loosing events

danni, 2019-05-02:
Possible. Happy to close this.

closed 2019-05-02 by danni