serialport/node-serialport #1838: Sometime first bytes are missing

phhoef, 2019-04-16:
### Summary of Problem
I am awaiting a `READY` sequence from the Arduino.
After opening the port, I send a `\0` for initialization. The Arduino answers with the `READY` sequence.
Sometimes, the serial port does not receive the `READY` seq.

(Please answer all 3)

- What are you trying to do?
I am building a small Electron app, to control my Arduino Due.
I've built a small command interface on the Due, that can set and get some variables and store them on a EEPROM.
\
I learned, that the serial port needs to run on the `main process` of Electron.
That's why I send IPC events from the `renderer process` to the `main process` and then send the command via serial port. After receiving data, I send them back to the `renderer process` via IPC events.
\
I am using the `Ready` Parser to recognize the init sequence.
When the Arduino is ready, I attach the `Delimiter` Parser. Each response from the Arduino is terminated by `?\r\n`.

- What happens?
In the UI of the app, I have a dropdown menu for selecting the port.
In addition, I've a connect button. When the connection is established there are happening a few things:
1. When the `open` event triggers, I send a `\0` for initialization.
2. When the `ready` event from the `Ready` Parser is triggered, the Arduino is successfully connected and the `Delimiter` Parser is attached
3. Immadately after the device is ready, the first command is send to get the firmware version of the Arduino.
\
The problem is, that sometimes, the `READY` sequence is not received.
I am quickly connecting and closing the connection and after 3-4 times it happens.
I am pretty sure, that the Arduino sends the `READY` sequence properly, for two reasons.
First, when doing the same manually with CoolTerm, I am getting the response reliable.
Second, I attached a logic analyzer and see, that the bytes are sent from the Arduino.

- What should have happened?
I would expect, that the `READY` sequence is always received.
Right now, the app gets stuck when this case occurs. The port is actually open, but I do not get the ready event.

### Code to Reproduce the Issue

```js
const serialport = require('serialport')
const Delimiter = require('@serialport/parser-delimiter')
const Ready = require('@serialport/parser-ready')
ipcMain.on('request-serial-disconnect', event => {
  if(arduinoPort && arduinoPort.isOpen) {
    arduinoPort.close();
  }

  delete arduinoPort;
});

let arduinoIsReady = false;
let arduinoPort;
ipcMain.on('request-serial-connect', async (event, device) => {
  arduinoIsReady = false;

  arduinoPort = new serialport(device, {
    baudRate: 9600,
    autoOpen: false
  });

  const ready = arduinoPort.pipe(new Ready({ delimiter: 'READY\r\n' }));
  
  ready.on('ready', () => { 
    console.log('Arduino is ready!')
    mainWindow.webContents.send('response-serial-connected', { success: true });
    arduinoIsReady = true;

    const parser = arduinoPort.pipe(new Delimiter({ delimiter: '?\r\n' }));
    parser.on('data', onData);
  });
  
  arduinoPort.on('error', onError);
  arduinoPort.on('open', () => {
    console.log('opened');
    arduinoPort.write('\0');
  }); // reset buffers)
  arduinoPort.open();
})

function onData(data) {
  let string = data.toString('utf-8').trim();
  console.log('Received ' + string);
  mainWindow.webContents.send('response-serial-command', string)
}

function onError(error) {
  console.log(error);
  mainWindow.webContents.send('response-serial-connected', { success: false, error: error.message });
}

ipcMain.on('request-serial-command', async (event, command) => {
  
  let start = Date.now();
  while(!arduinoIsReady && Date.now() - start <= 2000)
   await sleep(100);

  console.log(command);
  arduinoPort.write(command);
})

async function sleep(ms) {
  return new Promise(resolve => setTimeout(resolve, ms));
}
```

This is the response, I am logging with the code above:
```
opened
Arduino is ready!
GET:VAR_VERSION

Received VAR_VERSION:v4.0

opened
Arduino is ready!
GET:VAR_VERSION

Received VAR_VERSION:v4.0

opened
Arduino is ready!
GET:VAR_VERSION

Received VAR_VERSION:v4.0

opened
==> READY IS MISSING
```

### Versions, Operating System and Hardware

- SerialPort@? 7.1.4
- Electron 4.1.4
- Node.js v11.11.0
- Windows? Linux? Mac? macOS Mojave (10.14.4)
- Hardware and chipset? Arduino Due

reconbot, 2019-05-02:
When arduinos see the port open (DTR I think?) they reboot. Depending on when the `\0` gets sent the arduino may never see it. In general you have the arduino just send a "ready" event without being prompted

phhoef, 2019-05-02:
Thanks for your reply.
I've adapted the original boot loader of the Atmega 2560 on the Arduino Due to suppress the reboot.
The Arduino now only resets with baud rate 1200 (for flashing) and 1800 (my new magic baud rate for reset)

reconbot, 2019-05-06:
How does it detect baud rate changes?

phhoef, 2019-05-07:
I used the same mechanism as Arduino for the magic baud rate 1200.
https://github.com/arduino/Arduino/blob/976f7588b7f68bd3c5fce52c15ccfe738589085a/hardware/arduino/sam/firmwares/atmega16u2/arduino-usbserial/Arduino-usbserial.c#L271

I just introduced the 1800 baud and suppresses the reboot when connecting, no matter if DTR is high or low.

reconbot, 2019-05-07:
I wonder how that works? As far a I know from my research on the uart protocol, it has to receive data to figure any of that out. I wonder if it can always figure it out fast enough to preserve all the data. This is my first time seeing a device that can auto detect the baudrate.

phhoef, 2019-05-07:
Honestly, I don't know. You have by far more experience with all the UART stuff.
I just saw, that Arduino is doing this for the Due (I am using a Due) for bringing it in the flash mode. Arduino introduced the magic baud rate 1200 for that reason. I just adapted this mechanism.

I introduced a new flag for the 1800bps.
In the following code snippet, I am recognizing the 1800 baud. The 1200 baud are original code from Arduino.
```
/* Must turn off USART before reconfiguring it, otherwise incorrect operation may occur */
	UCSR1B = 0;
	UCSR1A = 0;
	UCSR1C = 0;

	long bps = CDCInterfaceInfo->State.LineEncoding.BaudRateBPS;
	Selected1200BPS = (bps == 1200);
	Selected1800BPS = (bps == 1800);

	UBRR1  = SERIAL_2X_UBBRVAL(bps);
	UCSR1C = ConfigMask;
	UCSR1A = (1 << U2X1);
	UCSR1B = (1 << RXCIE1) | (1 << TXEN1) | (1 << RXEN1);
```

And in this function I assure, that the reset timer is only set, when the 1800 baud occurred.
```
void EVENT_CDC_Device_ControLineStateChanged(USB_ClassInfo_CDC_Device_t* const CDCInterfaceInfo)
{
	PreviousDTRState = CurrentDTRState;
	PreviousRTSState = CurrentRTSState;
	CurrentDTRState = (CDCInterfaceInfo->State.ControlLineStates.HostToDevice & CDC_CONTROL_LINE_OUT_DTR);
	CurrentRTSState = (CDCInterfaceInfo->State.ControlLineStates.HostToDevice & CDC_CONTROL_LINE_OUT_RTS);

	if (Selected1200BPS) {
		/* Start Erase / Reset procedure when receiving the magic "1200" baudrate */
		ResetTimer = 120;
	} else if (Selected1800BPS) {
		/* Reset on rising edge of DTR */
		ResetTimer = 50;
	}
}
```

As far as I can tell, it's working pretty well.

I played around with the parsers and I think the bytes are lost here.
When just using the onData event without any parsers, the bytes seem to arrive.

reconbot, 2019-05-07:
Fair, it’s safe to pipe the ready parser to your delimiter parser, once the ready event fires it’s just a pass through for data events.

reconbot, 2019-05-16:
I think the ready parser will consume bytes that you wanted for the delimiter parser. I'm going to change my previous message to say "you should pipe your ready parser to the delimiter parser" or you will lose bytes.

closed 2019-05-16 by reconbot