serialport/node-serialport #1840: TypeScript definition for SerialPort.MockBinding should not be readonly

tagyoureit, 2019-04-17:
The documentation for testing serialport [here](https://serialport.io/docs/en/api-binding-mock) and [here](https://serialport.io/docs/en/guide-testing) say to use `SerialPort.Binding = MockBinding`.

However, the TypeScript definition files has this as a readonly property: `static Binding: SerialPort.BaseBinding;`

In order to set SerialPort.Binding to MockBinding TypeScript needs a property that can be changed.

[image]
[image]

tagyoureit, 2019-04-19:
nvm.

closed 2019-04-19 by tagyoureit

reconbot, 2019-04-19:
@tagyoureit did someone update the typescript project or did you find another way?

tagyoureit, 2019-04-19:
I'm not sure... I'm not getting the error anymore.

reconbot, 2019-04-19:
🤔