serialport/node-serialport #1841: \r implicitely added

fralbo, 2019-04-18:
### Summary of Problem

I made  a first node serial installation few weeks ago (don't really remember if I made it by npm or downloading the archive) and made a successfull try.

Yesterday I installed it on another PC with npm and I saw that my exact same code didn't work anymore and I discovered with a spy tool that a '\r' is implicitely added (after the '\n' I add myself) at the end of the written string.

Does it tell you something?

### Code to Reproduce the Issue

```js
const serialport = new SerialPort('COM6')

serialport.write('?R00000005;51\n');

### Versions, Operating System and Hardware

- SerialPort@7.1.4
- Node.js v10.15.3
- Windows 10
- Hardware and chipset? FTDI

fralbo, 2019-04-18:
In fact it doesn't depends on the installation way.

I have to files which seem to do the same thing and one works not the other and I currently cannot figoure out what.

reconbot, 2019-04-18:
Is there an error? Is the version of node the same?

closed 2019-07-08 by stale[bot]