serialport/node-serialport #1842: SerialPort list function works only the first time

LucaSoldi, 2019-04-19:
Hi,
I have a problem with listing function, at the app startup it works fine. I cannot update the com list after the first scan. 
My project is based on electron, I used the example here (https://github.com/nodebots/electron-serialport) and even in this project the polling loop doesn't work.
Any suggestion?
Thanks!

### Code to Reproduce the Issue
```
const serialport = require('serialport')

setInterval(function() {
  console.log("list");
  serialport.list((err, ports) => {
    console.log('ports', ports);
  })
},
3000);

// Code
```

### Versions, Operating System and Hardware

- SerialPort@?^7.1.4
- Node.js v? 8.14.0
- Mac

reconbot, 2019-04-19:
This sounds like https://github.com/node-serialport/node-serialport/issues/1794 should go away if you upgrade your xcode and cli tools.

LucaSoldi, 2019-04-23:
I'm using Mac OSX High Sierra, and the update of cli tools not solve the problem :(

reconbot, 2019-04-23:
Can you try the code change suggested in the other issue?

LucaSoldi, 2019-04-23:
I update my os Mojave and reinstall all my electron app dependencies, now it works :)
Thanks @reconbot!

closed 2019-04-23 by LucaSoldi