serialport/node-serialport #1847: A single changelog

reconbot, 2019-04-27:
## 💥 Proposal

### What feature you'd like to see
I'd like to see a single global changelog for the project.

## Motivation
The current changelogs aren't very helpful for understanding what changed when you upgrade. Dependent packages just say "version bump".

## Pitch

Figure out how lerna generates these and see if it's got a better option.

tkurki, 2020-06-08:
https://github.com/lerna/lerna-changelog apparently works on labeled PRs, so it would be a good idea to start labeling PRs so that eventually it will have something to chew on. And labels would be useful for hu-mans also.

The default labels are
- breaking (💥 Breaking Change)
- enhancement (🚀 Enhancement)
- bug (🐛 Bug Fix)
- documentation (📝 Documentation)
- internal (🏠 Internal)

GazHank, 2021-07-10:
Hi @tkurki if we add these as labels, would I be correct to assume that this would just work going forward? That is to say reclassifying previous PRs wouldn't really help us?

tkurki, 2021-07-19:
Sorry, I have not looked into how this project creates releases and I don't know if you can update just the notes for existing Github releases.

I just think it would make sense to
- pick a tool for generating release notes from commits or PRs
- start applying the mechanism (naming, GH labels, etc) that the tool uses
- add release notes generation to the release process

GazHank, 2021-07-19:
A few observations so far while trying this:

- We will probably need to reclassify some PRs, this items is a good example, while adding this could be considered an enhancement, for most users this would probably just be an internal change
- we will need to see how well the lerna change log works with the overall package structure (https://github.com/lerna/lerna-changelog/issues/26)
- While it seems to work for unreleased info (so could be used just before publication), the historic functioanlity doesnt seem to work (for me)

GazHank, 2021-07-19:
I've drafted https://github.com/serialport/node-serialport/pull/2298 for now

I've kept this pretty close to the current change log, and we can drop the commiters info if we like (https://github.com/lerna/lerna-changelog#configuration) .

Based on this initial config at the moment it generates:

## Unreleased (2021-07-19)

#### Chores
* [#2294](https://github.com/serialport/node-serialport/pull/2294) Use Dependabot to keep node-abi up to date ([@GazHank](https://github.com/GazHank))

#### Committers: 1
- Gareth Hancock ([@GazHank](https://github.com/GazHank))