serialport/node-serialport #1853: Build for Windows on Mac using electron-builder

DonaldChiang, 2019-05-02:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
I am trying to build a multiple platform electron app with node-serialport on Mac
- What happens?
Electron can run on Mac perfectly but on Windows it shows this error 
```
A JavaScript error occurred in the main process

Uncaught Exception:
Error: \\?\C:\Users\MyUser\AppData\Local\Temp\4deb32d9-50e4.....bf7b.tmp.node is not a valid Wind32 application.
```

[image]

And I am using electron-builder to build my electron app with no errors and I already run electron-rebuild every time before building.

- What should have happened?
It should work as Mac's version

### Code to Reproduce the Issue

```js
"serialport": "6.2.2"
"electron": "4.0.5"
"electron-builder": "20.34.0"
"electron-rebuild": "^1.8.4"
```

### Versions, Operating System and Hardware

- SerialPort@?
6.2.2
- Node.js v?
v8.12.0
- Windows? Linux? Mac?
Build on macOS 10.14.3 Mojave
- Hardware and chipset? (Prolific/FTDI/Other)

iamsush, 2019-05-02:
Is it working fine before packaging it on windows?

DonaldChiang, 2019-05-02:
Thanks for reply. When I build electron on Windows it works fine. I am trying to use prebuild-install before build on Mac now but can't figured out some step, still working on it.

iamsush, 2019-05-02:
Your concern is not clear to me. Could you please elaborate your issue?

DonaldChiang, 2019-05-02:
@iamsush OK, I will create a repository for reproduction on Github, thanks.

iamsush, 2019-05-02:
Ok, thanks.

DonaldChiang, 2019-05-02:
Here is the testing repository: https://github.com/DonaldChiang/node-serialport-testing
1. clone this repository
2. run yarn install
3. run yarn electron:build:windows on Mac
4. copy node-serialport-testing Setup 0.1.0.exe to Windows and install

And you can see this error

Thanks!

iamsush, 2019-05-02:
Even I have had this issue, to resolve it I established my code on windows machine and it worked fine. 
(Quite possible because for windows, it needs windows-build-tools and other things, I am not sure of the cause but this resolves it)

Have a look at this issue, [#966](https://github.com/node-serialport/node-serialport/issues/966). Does this resolve yours?

DonaldChiang, 2019-05-02:
Yes, I can build electron on Windows and it works fine on Windows. So now my question is:
Can I compile node-serialport on Mac(or Linux) that is for Windows? Or use prebuild to do this?

it look like I can only compile node-serialport on Windows if I want to build an electron app for Windows version

Is there any guide for setup prebuild or node-gyp for multiple platform build?

Thank you very much!

iamsush, 2019-05-02:
Even I am not sure of this, Let @reconbot reply to this.

reconbot, 2019-05-02:
I know there are electron build tools for cross platform building. (I have
in the past built Linux windows and OS X from my OS X laptop) they wouldn’t
be serialport specific.

Josverl, 2019-05-02:
Hi pitching in cause im also interested in bundling for multiple platforms / electron 

I would expect that there is a way to use the precompiled binaries that are located here :
https://github.com/node-serialport/node-serialport/releases 
and therefore avoid the need for cross-compilers or running the builds on multiple platforms.

That is also mentioned in the docs 'We use prebuild to compile and post binaries of the library for most common use cases (Linux, Mac, Windows on standard processor platforms). '

But I have not been able to find clear documtation on how to make use of the posted binaries.

is my understanding correct, and if so , can you post a sample of how to use the prebuilts  ?

reconbot, 2019-05-02:
It’s used by the npm install process, we use the package named
prebuild-install to download them. And the package named bindings to load
them. I have no idea how to distribute electron apps without building them.
-- 
---
Francis Gulotta
wizard@roborooter.com

danni, 2019-07-24:
I've run into this since trying to upgrade to Electron 5, for which there are no prebuilt packages. It looks like something between electron-builder calling node-gyp-rebuild and @serialport/bindings, something is not getting compiled.

https://github.com/electron-userland/electron-builder/issues/3660 has some details, e.g. here's the result of rebuilding:

```
-rwxr-xr-x  1 danni  staff  201072 19 Jun 13:02 usb/build/Release/usb_bindings.node
-rwxr-xr-x  1 danni  staff  885752 19 Jun 13:02 iltorb/build/Release/iltorb.node
-rwxr-xr-x  1 danni  staff  55244 **28 Apr 00:58** @serialport/bindings/build/Release/bindings.node
```

https://github.com/electron-userland/electron-builder/issues/4012 has a debug log:

```
  • rebuilding native dependencies  dependencies=@serialport/bindings@2.0.8, iltorb@2.4.3, usb@1.6.0 platform=darwin arch=x64
  • map async       taskCount=3
  • rebuilding native dependency  name=@serialport/bindings version=2.0.8
  • rebuilding native dependency  name=iltorb version=2.4.3
  • execute command  command=/usr/local/Cellar/node/12.3.1/bin/node /usr/local/Cellar/yarn/1.16.0/libexec/bin/yarn.js run install workingDirectory=/Users/danni/src/SPT/spt-bridge/node_modules/iltorb
  • execute command  command=/usr/local/Cellar/node/12.3.1/bin/node /usr/local/Cellar/yarn/1.16.0/libexec/bin/yarn.js run install
                     workingDirectory=/Users/danni/src/SPT/spt-bridge/node_modules/@serialport/bindings
  • command executed  executable=/usr/local/Cellar/node/12.3.1/bin/node out=yarn run v1.16.0
$ prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
Done in 0.27s.
```

marceloch2, 2019-08-28:
I have the same issue with Electron 6, did you solve it @danni ?

danni, 2019-08-28:
I haven't come back to it yet, sorry.

danni, 2019-10-28:
I don't know if anything changed in the most recent update, but the most recent update permitted me to upgrade to Electron 6.

closed 2020-01-03 by stale[bot]