serialport/node-serialport #1854: Not able to receive data after resetting esp32 on windows

iamsush, 2019-05-02:
### Summary of Problem

I am trying to read data from esp32. It works as expected on Linux but does not behave the same on windows. In the starting, esp32 on both the platform (Linux and windows) shows correct data but on resetting it does not.
(Sample code 'GetChipID.ino' is uploaded from arduino IDE to esp32)

=> On linux, 
On resetting esp32, below is the output,
```
ESP32 Chip ID = 54397BA4AE30
ESP32 Chip ID = 54397BA4AE30
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10208
load:0x40080400,len:6460
entry 0x400806a4
ESP32 Chip ID = 54397BA4AE30
ESP32 Chip ID = 54397BA4AE30
ESP32 Chip ID = 54397BA4AE30
```
=> On Windows, , below is the output, (it stops receiving data after resetting esp32)
```
ESP32 Chip ID = 54397BA4AE30
ESP32 Chip ID = 54397BA4AE30
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

### Code to Reproduce the Issue

```
const SerialPort = require('serialport');
const Readline = require('@serialport/parser-readline');
const port = new SerialPort('/dev/ttyUSB0', // Port as per usage
{
    baudRate: 115200
});
const parser = port.pipe(new Readline());

parser.on('data', data => console.log(data));
```

### Versions, Operating System and Hardware

=>Windows 8/10, 
- SerialPort@7.1.5
- Node.js v10.15.3
- Hardware = ESP32

=>Linux,
- SerialPort@7.1.5
- Node.js v8.16.0
- Hardware = ESP32

reconbot, 2019-05-04:
How are you connected to the ESP32?

iamsush, 2019-05-07:
I am using generic dev module of espressif connected via Micro USB cable.

reconbot, 2019-05-07:
Can you run both with debug mode https://serialport.io/docs/en/guide-debugging And share the output here?

iamsush, 2019-05-07:
Output did not change much with debug mode, Am I doing something wrong?
Below, is the output,
```
set DEBUG=serialport/* & node serialport.js
[1] 44772
ESP32 Chip ID = F4E46BA4AE30
ESP32 Chip ID = F4E46BA4AE30
ESP32 Chip ID = F4E46BA4AE30
ESP32 Chip ID = F4E46BA4AE30
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

reconbot, 2019-05-07:
Yeah it’s not seeing the env var. try `DEBUG=*` ?

We use the npm package debug for this https://www.npmjs.com/package/debug#windows-command-prompt-notes

reconbot, 2019-05-07:
Ohhh you’re using the windows env var setting for Linux.

iamsush, 2019-05-07:
I am doing this in windows as mentioned in the docs.
In linux, I have no issues.

reconbot, 2019-05-07:
Well.. there are no errors, the read is pending, which means we've asked the operating system for more data. I'm not sure. Do you know how to see if the ESP32 removes itself and adds itself to device manager or something?

iamsush, 2019-05-08:
I am not sure how does that work.
Also, I would like you to know that if I am uploading the code from arduino IDE and checking the data in its Serial Monitor, everything works fine.

The problem is there when I upload the code from arduino IDE and check the data in my code (where I use serialport module for interfacing).

Can you figure out something with this information?

iamsush, 2019-05-13:
@reconbot, Could you please tell me how can I proceed with this?

reconbot, 2019-05-13:
I don't know @iamsush it starts to read and doesn't fail but gets no data from the operating system. I don't have an esp32 to test with.

iamsush, 2019-05-15:
I have tried with other development boards and they are working perfectly fine with _serialport_. I eagerly want to work with _esp32_ using _serialport_ module.
Let me know how can I help you with this. I would be really glad to do so.

zrecore, 2019-05-16:
What's the baudrate at?

iamsush, 2019-05-16:
@zrecore, the communication is set at 115200 Baudrate.

reconbot, 2019-05-16:
https://twitter.com/tobozotagada/status/1129088947546730499?s=21

tobozo, 2019-05-16:
long story short : resetting isn't required to just "talk" to the serial (opening the port is enough), and any attempt to flash the esp32 should expect "waiting for download" before sending binary data

iamsush, 2019-05-17:
Yes I understand that resetting is not required for connecting to serial port. 
Once I get connected to serial port I am getting data correctly until I press reset button on ESP32 dev board.  As soon as I press reset button it goes in "waiting for download mode" i.e boot mode, but actually it should remain in normal SPI flash mode. After uploading code through above stated method if I open Serial Monitor in Arduino IDE and press reset button it works fine and my ESP32 remains in normal flash mode only.

tobozo, 2019-05-17:
@iamsush then could be the differences between Windows and Linux ways to handle the serial driver

speculation: Windows triggers the reset when the driver is loaded in user mode while Linux does this in privileged mode and thus does not trigger the reset ?

suggestion: have you tried a different version of your Windows cp21x/ch34x driver ?

iamsush, 2019-05-20:
@tobozo, I don't think this will change anything much. I tried with the latest CP210x drivers on Windows 10 as well as 8 but that seems to make no difference. Even for CH340, I used the recent and most commonly used drivers, issue still remains.

tobozo, 2019-05-20:
@iamsush sometimes *downgrading* CP210x may give different results too (at least it did on my M5Stack rev 1). 

Have you succeeded into flashing your esp32 on windows outside nodejs ? If so what com port was it using, what was the associated driver ?

iamsush, 2019-05-21:
@tobozo, I don't think the issue is caused by the drivers as everything works fine with Arduino IDE on the same machine.

tobozo, 2019-05-21:
I guess it's time to show the JS code you're using then :)

iamsush, 2019-05-21:
@tobozo, I have already shared the code in the issue. Please let me know if anything else is required.

tobozo, 2019-05-21:
does it improve with the following code example ? 

```
const SerialPort = require('serialport')
const parsers = SerialPort.parsers

// Use a `\r\n` as a line terminator
const parser = new parsers.Readline({
  delimiter: '\r\n',
})

const port = new SerialPort('/dev/ttyUSB1', {
  baudRate: 115200,
})

port.pipe(parser)
port.on('open', () => console.log('Port open'))
parser.on('data', console.log)
port.write('help\n')

// The parser will emit any string response
```

iamsush, 2019-05-22:
I tried to run the above script. It doesn't make any difference.
The issue is only after resetting esp32, when serialport stops reading data and goes into "waiting for download mode" i.e boot mode. If I don't reset esp32, everything works fine for me.

tobozo, 2019-05-22:
Found [this](https://www.esp32.com/viewtopic.php?t=334&start=10#p1497) as an beginning of explanation.

> On Windows there is a 4ms delay between when "EN" goes high and when "IO0" goes low. This is caused by the way the auto-reset circuit is wired, to keep compatibility with "normal" serial programs (there is a small table in the schematic that shows how DTR/RTS control lines related to the EN/IO0 pins). The value of "IO0" seems to be being read/latched during this 4ms period.

A quick test would confirm this by adding a 10k pullup on IO0, if this improves your situation then this corroborates my first hint : downgrade your driver and hope that older versions don't have this delay.

@reconbot can anything be done at library level ? esptool (uses python) does not seems to be affected by this issue.

reconbot, 2019-05-23:
Maybe but without knowing what is different it’s hard to isolate lines of code. We both call the same c code under the hood.

That privileged vs not privileged idea is interesting. I always thought if you could read/write you had full access to all functions. But I can’t really confirm my assumptions.

iamsush, 2019-05-25:
@tobozo , I observed that the IO0 pin always remain low, and pulling it up with 10K resistor didn't change the state of the pin , it still remained low. Now as this pin remains low, the ESP goes in boot mode on every reset. This pin goes low as soon as I open the Serial port with my script.

tobozo, 2019-05-25:
this is how esptool does it, maybe this can be implemented in the JS code ?

[image]

tobozo, 2019-05-25:
BTW there's this package:

https://www.npmjs.com/package/esptool

iamsush, 2019-05-29:
@tobozo, The last update to the [package](https://www.npmjs.com/package/esptool) was made 4 years ago, I am not sure if I should depend on it.
@reconbot, Is there anything that we can try on this?

tobozo, 2019-05-29:
@iamsush looks like you need to finish the work on this package by implementing esptool logic by yourself, or give up serialport and use esptool.py from a child-process

iamsush, 2019-05-29:
@reconbot, I observed that DTR pin goes low when I open the port via serialport module, which is contradictory to the way Arduino IDE handles opening of the port (DTR pin remains high). 
And as per the auto reset logic circuit of ESP32, if DTR is low and RTS is High, the gpio 0 goes low and hence, on continuous reset, ESP32 goes in boot mode.

Also, while I was going through the node-serialport codebase, I observed that DTR state (in [windows](https://github.com/serialport/node-serialport/blob/master/packages/bindings/src/serialport_win.cpp)) is controlled via hupcl option. And just to try I included hupcl option to false in my script, which resolved my issue.
But I am not sure what _hupcl_ does, Could there be any new issue if I make it to _false_? 
Is this [issue](https://github.com/serialport/node-serialport/issues/1678) related?

iamsush, 2019-06-07:
@reconbot. any help?

reconbot, 2019-06-07:
It’s a good discovery, I need to investigate it. 

(Today is my last day at my day job so I’ll have a bit of time to concentrate here.)

iamsush, 2019-06-07:
@reconbot, Ok sure. Thanks though :)

danni, 2019-06-13:
Possible relationship with #1836 ?

closed 2019-08-19 by stale[bot]