serialport/node-serialport #1858: XMODEM?

bdmfab, 2019-05-03:
Is there currently any way to implement xmodem protocol?

reconbot, 2019-05-03:
I see some npm packages but haven’t used any of them.

bdmfab, 2019-05-03:
Thanks for the reply. I saw that as well, it wasn't apparent if you could
pipe through. It requires some ack nak callbacks as well.

reconbot, 2019-05-04:
The npm package `serialport` does make use of nodejs streams which work the same as a tcp socket that at least one of the packages use. So there's a good chance it might work.

reconbot, 2019-05-20:
Let us know if you're able to make anything work.

closed 2019-05-20 by reconbot

bdmfab, 2019-07-03:
I managed to make xmodem work and have tested it quite extensively. Here is
the sample code, let me know if you have any questions.

-- 
Thanks,

Dan

const fs = require('fs');
const SerialPort = require('serialport');
const xmodem = require('xmodem.js');
const Net = require('net');
const EOT = 0x04;
const STRT = 0x11;
let xmod_soc = {};

const buf = fs.readFileSync('./sampletext.txt');

const port = new SerialPort("/dev/ttyUSB0", {
    autoOpen: true,
    baudRate: 38400,
    dataBits: 7,
    parity: "even",
    stopBits: 1,
    xon: false,
    xoff: false,
    rtscts: false
});

port.on('open', () => {
  console.log("Serial Port Opened");          
});

port.on('data', (data) => {
  console.log('Data - ' + data);
  xmod_soc.write(data);  
});

port.on('close', () => {
  console.log('Port Closed');
});

port.on('error', (err) => {
  console.error('Port Error - ' + err);
});

const server = Net.createServer((sock) => {
    //on connected
    xmod_soc = sock;
             
    sock.on('data', (data) => { port.write(data); });
  
    sock.on('end', () => { console.log('client disconnected') });    
});
  
server.on('error', (err) => {
    throw err;
});

server.on('close', () => {
  console.log('Server Closed');
  port.close();
});

server.listen(1337, '127.0.0.1');

let xmod = new Net.Socket();

xmod.connect(1337, '127.0.0.1', () => {
	console.log('xmod connected');	
});

xmod.on('data', (data) => {
  console.log('xmod Received: ' + data);	
});

xmod.on('close', () => {
  console.log('xmod connection closed');
  server.close();	
});

xmodem.on('ready', (count) => {
  console.log('Xmodem ready with ' + count + ' packets.'); 
  // Notify external device we want to use xmodem protocol
  port.write('<Xmodem ini string>' + '\r\n');       
});

xmodem.on('start', (mode) => {
  console.log('Xmodem started - Mode = ' + mode);
});

xmodem.on('status', (status) => {
  console.log("Status - " + status);
  if (status.signal == 'EOT') {
    xmod.end();    
  }
});

function x_send() {
  xmodem.send(xmod, buf);
};

setTimeout(x_send, 200);