serialport/node-serialport #1860: Nanoserver support

desertspider, 2019-05-04:
### Summary of Problem

- What are you trying to do?
Running zigbee2mqtt on nanoserver
- What happens?
Zigbee2mqtt crashes when connection to the zigbee stick
- What should have happened?
no crashes.

Thing this topic answers all questions about my problem;
https://github.com/Koenkk/zigbee2mqtt/issues/1479

In short: zigbee2mqtt, bases on node.js with serialport does not work on nanoserver, but does work on servercore. Serialport probably uses an api not available on nanoserver. Question is. can serialport be fixed to be used on nanoserver, or what libraries should i sideload on nanoserver to make serialport work?

### Versions, Operating System and Hardware

- Node.js v node-v10.15.3-win-x64/node-v11.15.0-win-x64. 
- Windows

reconbot, 2019-05-04:
What is nanoserver? What is servercore?

desertspider, 2019-05-04:
> What is nanoserver? What is servercore?
Servercore:
https://docs.microsoft.com/nl-nl/windows-server/get-started/getting-started-with-nano-server
"This is a base image for Windows Server containers. This image carries the Windows Server Core base OS image."
aka Full windows server without GUI

Nano server:
https://docs.microsoft.com/nl-nl/windows-server/get-started/getting-started-with-nano-server
aka nano size version of full server version. 
It has a lite version of WIN32, with legacy API's removed.

reconbot, 2019-05-05:
We don’t use the uwp apis if that’s what it requires.

desertspider, 2019-05-06:
> We don’t use the uwp apis if that’s what it requires.

Well, for it to work on Nanoserver, it does.

reconbot, 2019-05-06:
I think there’s some sort of shim you can add or compile in. Visual studio used to do that for windows 10 iot.

desertspider, 2019-05-06:
Ok loaded Kernel32.dll and setupapi.dll. Got this error now;

```
  0: ExitFrame [pc: 0000023EAB7D0481]
Security context: 0x03fb16b1d9f1 <JSObject>
    1: debug [0000033DB26EC771] [C:\app\node_modules\cc-znp\node_modules\debug\src\common.js:~67] [pc=0000023EAB7ED49B](this=0x03a7e9604981 <JSGlobal Object>)
    2: arguments adaptor frame: 1->0
    3: close [000001A812975A69] [C:\app\node_modules\cc-znp\node_modules\@serialport\stream\stream.js:424] [bytecode=000002762F45B589 offset=47](this=0x03ad7aa295...
```

reconbot, 2019-05-06:
I don’t know enough about windows to help here. I know for docker alpine and Ubuntu both work really well.

desertspider, 2019-05-08:
To bad. Thanks for your help. Reason why i have to use Windows docker container is that com device pass-through is not supported to Linux containers on windows. I will use the full server core version for now.

closed 2019-05-08 by desertspider

reconbot, 2019-05-08:
That’s a good reason.